stephane/libmodbus #381: Windows/Cygwin: ./configure packaged with versions 3.0.6 and 3.1.4 produces invalid Makefile

R030t1, 2017-05-02:
## libmodbus version

  3.0.1, 3.1.4; Git `HEAD` fails with a syntax error in `autogen.sh`.

## OS and/or distribution

  Windows with Cygwin.

## Environment

  x86_64.

## Description

  For 3.1.4:
  `Makefile:284: *** missing separator. Stop.`

  For 3.0.6:
  `Makefile:266: *** missing separator. Stop.`

## Expected behavior

  Successful compilation.

## Actual behavior

  Make is unable to interpret the makefile.

## Steps to reproduce the behavior (commands or source code)

  `./configure && make`

stephane, 2017-05-02:
Makefile isn't provided by git repo. Did you run ./autogen.sh to generate it from your git checkout?

R030t1, 2017-05-02:
Sorry, do you not support your releases from this repository? I filed this in response to the inability to compile both releases offered on the download page. If you don't want issues with the releases here I can update this bug to be about the issue with `autogen.sh`, but I strongly suspect the issues are related.

To answer your immediate question I am unable to run `autogen.sh` due to a syntax error (unexpected end of file). Seeing as the error with the Makefile has persisted between two releases I am going to hazard a guess that it is still present. I can't see anything obviously wrong with the autogen script.

stephane, 2017-05-05:
Yes, of course, I try to support at least the recent releases but`Makefile` files aren't provided by the release tarballs so my first idea is just to suspect an issue with your env. Each generated Makefile could be different (OS, version dependent), saying:

`Makefile:284: *** missing separator. Stop.`

doesn't help me to solve your issue.

BTW it's first time in more than 10 years I received a bug report about a broken `Makefile`.
May be you disk is full, your RAM broken, ...
I can't help you with (unexpected end of file).

Have you been able to untar the release w/o issue?

```
md5sum libmodbus-3.1.4.tar.gz
b1a8fd3a40d2db4de51fb0cbcb201806  libmodbus-3.1.4.tar.gz
```

R030t1, 2017-05-05:
Sorry, I didn't mean to claim that I was using a provided Makefile. For the releases cited (I'm more interested in the 3.1.4 development version) there is no `autogen.sh` in the produced directory structure. The release ships a configuration script which I run to produce the Makefile.

I verified the MD5 sum and found it matched without redownloading. When I attempted to build it this time I found that there were no errors in the Makefile, and that having MinGW in my PATH seemed to be confusing Cygwin and the configure script. I removed it and now the project builds.

closed 2017-05-05 by R030t1