stephane/libmodbus #383: modbus-data.c: bswap checking inconsistency

dreamflow, 2017-05-04:
in [src/modbus-data.c](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c) in [line 57](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c#L57)

    #if !defined(__CYGWIN__) && !defined(bswap_16)

and [line 65](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c#L65)

    #if !defined(bswap_32)

look inconsistent.

shouldn't both lines be more consistent ?
like for example adapt [line 65](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c#L65) according to [line 57](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c#L57) and change [line 65](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c#L65) to

    #if !defined(__CYGWIN__) && !defined(bswap_32)

or vice versa adapt [line 57](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c#L57) according to [line 65](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c#L65) and change [line 57](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c#L57) to

   #if !defined(bswap_16)

stephane, 2017-05-05:
I can't remember why there is a check against CYGWIN on [line 57](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c#L57).

dreamflow, 2017-05-08:
i'm in favour of merging the [bswap-check](https://github.com/stephane/libmodbus/tree/bswap-check) branch into the [master](https://github.com/stephane/libmodbus) branch.

the  CYGWIN check in [src/modbus-data.c](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c) in [line 57](https://github.com/stephane/libmodbus/blob/master/src/modbus-data.c#L57) is not needed
and it makes not really sense, because the bswap_16 function
is not provided by CYGWIN. the bswap_16 function is provided
by gcc as __builtin_bswap16 . ggc provides its __builtin_bswap16
independet of CYGWIN, for example when MSYS is used instead of CYGWIN.

commit-history of the CYGWIN-check:
created/added in commit https://github.com/stephane/libmodbus/commit/cb4d3ea643be4e7bf047d634577b6259fa305574
improved in commit https://github.com/stephane/libmodbus/commit/67cacf8c54b41bf0f6b62398e75fe518505f964c

stephane, 2017-06-21:
Merged. Thank you.

closed 2017-06-21 by stephane