stephane/libmodbus #384: Maximum number of coils reading and writing

esc39, 2017-05-09:
Hi everybody,

Firstly, I would like to say many thanks to the library creators and to all their contribuators. I am using this library and I am checking the maximum value for the reading and the writing. I have already done this job for the registers and it worked well. However, it is not working with the coils.
So, it is working with the read and write of 200 coils but not with the maximum values defined in the standard modbus protocol and also given in the modbus.h file :
#define MODBUS_MAX_READ_BITS              2000
#define MODBUS_MAX_WRITE_BITS             1968
Someone have an idea why it is not working with the maxium values ? Also, I tried with less than these values (1999 for the reading and 1967 for the writing) but I had the same results.

Many thanks for your help,
cesc39

karlp, 2017-05-09:
Are you sure your device actually allows reading that many coils at once?  Many devices have their own limits that can be substantially less than what the protocol supports.

esc39, 2017-05-10:
Hi,

Thanks for your answer. I am not sure I thought all devices followed the modbus standard protocol maybe not ... I am going to take a look. I will come back to you to tell you whatever the answer.

Thanks,
cesc39

esc39, 2017-05-10:
Hi,

You had right it is a device limit.

Thanks for your help,
esc39

closed 2017-06-21 by stephane