stephane/libmodbus #393: Timeout not respected

FabioGenesy, 2017-07-24:
Hi,
I use the stable version of libmodbus (3.0.6)  but I noticed a problem.
If a set a timeout passing a timeval to the modbus_set_response_timeout,
like 10ms, it's not respected but the program is blocked always for a second.
Are you aware of this problem?
I use the library for modbus TCP connection on Ubuntu 16.04

Thank you very muck!

stephane, 2017-07-24:
No issues are known and we have a few tests for that.
Are you able to test with latest version and/or provide your code?

FabioGenesy, 2017-07-24:
Hi, I replicate the problem in an example, developed with Qt, that I 
attached to this email.
Unfortunately I am forced to use version 3.0.6 and can not replace it.
The test is executed on Ubuntu 16.04

Thank you in advace

stephane, 2017-07-24:
You can't attach files to github issue. Please provide a very minimal extract of your code we can compile if you want help.

stefanofiorentino, 2017-07-24:
@FabioGenesy maybe you are passing wrong number format, did you check the
function signature and timeout type?

FabioGenesy, 2017-07-25:
Hi Stefano,
this is the code for initialize the timeout:

     struct timeval response_timeout;
     response_timeout.tv_sec = 0;
     response_timeout.tv_usec = timeOut*1000;
     modbus_set_response_timeout(m_modbus, &response_timeout);

where timeOut is an integer that represents the timeout in milliseconds, 
for example 10.
I tried also to set the tv_sec element but there is always the same problem.

-- 
Ing. Fabio Benevento
Genesy SrL
Via Malasoma, 24
56121 Pisa
tel 050 981970 - int. 203
fax 050 988138

________________________________________________________
Importante: Questo messaggio e gli eventuali allegati sono confidenziali
e riservati. Se vi è stato recapitato per errore e non siete fra i
destinatari elencati, siete pregati di darne immediatamente avviso al
mittente. Le informazioni contenute non devono essere mostrate ad altri,
né utilizzate, memorizzate o copiate in qualsiasi forma.

FabioGenesy, 2017-07-25:
I upload it with WeTransfer. This is the link:
https://we.tl/L1ocY8H3vA

Anyhow, this is the code for my modbus adapter:

modbus_adapter.h

#ifndef MODBUSADAPTER_H
#define MODBUSADAPTER_H

#include <QObject>
#include <QTimer>
#include <modbus/modbus.h>

/* Modbus function codes */
#define MODBUS_FC_READ_COILS                0x01
#define MODBUS_FC_READ_DISCRETE_INPUTS      0x02
#define MODBUS_FC_READ_HOLDING_REGISTERS    0x03
#define MODBUS_FC_READ_INPUT_REGISTERS      0x04
#define MODBUS_FC_WRITE_SINGLE_COIL         0x05
#define MODBUS_FC_WRITE_SINGLE_REGISTER     0x06
#define MODBUS_FC_READ_EXCEPTION_STATUS     0x07
#define MODBUS_FC_WRITE_MULTIPLE_COILS      0x0F
#define MODBUS_FC_WRITE_MULTIPLE_REGISTERS  0x10
#define MODBUS_FC_REPORT_SLAVE_ID           0x11
#define MODBUS_FC_MASK_WRITE_REGISTER       0x16
#define MODBUS_FC_WRITE_AND_READ_REGISTERS  0x17

class ModbusAdapter : public QObject
{
     Q_OBJECT
public:
      explicit ModbusAdapter(QObject *parent = 0);

      void modbusConnectRTU(QString port, int baud, QChar parity, int 
dataBits, int stopBits, int timeOut=1);
      void modbusConnectTCP(QString ip, int port, int timeOut=1);
      void modbusDisConnect();

      bool isConnected();

      void setSlave(int slave);
      void setFunctionCode(int functionCode);
      void setStartAddr(int addr);
      void setNumOfRegs(int num);
      void addItems();

      int packets();
      int errors();

      int modbusReadData(int slave, int functionCode, int startAddress, 
int noOfItems, uint8_t *dest);
      int modbusWriteData(int slave, int functionCode, int startAddress, 
int noOfItems, uint8_t *src);

private:
      QString stripIP(QString ip);
      modbus_t * m_modbus;
      bool m_connected;
      int m_ModBusMode;
      int m_slave;
      int m_functionCode;
      int m_startAddr;
      int m_numOfRegs;
      int m_packets;
      int m_errors;
      int m_timeOut;

public slots:
     void resetCounters();
[... 310 lines cut ...]
             i++;
         }
         return res;
     }
     else
         return "";

}

Thank you very much!

-- 
Ing. Fabio Benevento
Genesy SrL
Via Malasoma, 24
56121 Pisa
tel 050 981970 - int. 203
fax 050 988138

________________________________________________________
Importante: Questo messaggio e gli eventuali allegati sono confidenziali
e riservati. Se vi è stato recapitato per errore e non siete fra i
destinatari elencati, siete pregati di darne immediatamente avviso al
mittente. Le informazioni contenute non devono essere mostrate ad altri,
né utilizzate, memorizzate o copiate in qualsiasi forma.

FabioGenesy, 2017-07-28:
Have you been able to replicate the issue?

-- 
Ing. Fabio Benevento
Genesy SrL
Via Malasoma, 24
56121 Pisa
tel 050 981970 - int. 203
fax 050 988138

________________________________________________________
Importante: Questo messaggio e gli eventuali allegati sono confidenziali
e riservati. Se vi è stato recapitato per errore e non siete fra i
destinatari elencati, siete pregati di darne immediatamente avviso al
mittente. Le informazioni contenute non devono essere mostrate ad altri,
né utilizzate, memorizzate o copiate in qualsiasi forma.

jcwren, 2017-09-28:
@FabioGenesy, your WeTransfer page has been deleted. 

You can attach files to issues. See https://help.github.com/articles/file-attachments-on-issues-and-pull-requests/

closed 2017-10-25 by stephane