stephane/libmodbus #403: Set response timeout causes segmentation fault

IlhanSeven, 2017-09-26:
Platform: Beaglebone Black Angstrom linux
libmodbus: latest version

ctx = modbus_new_rtu("/dev/ttyO4", 9600, 'N', 8, 1);
    modbus_rtu_set_serial_mode(ctx, MODBUS_RTU_RS485);
    modbus_set_slave(ctx, 2);
    modbus_set_debug(ctx, 1);
    modbus_set_response_timeout(ctx, 0, 200000);

Commenting out the last line -> no seg fault.

What could be the reason?
Thanks for this work. It is quite useful.
Ilhan

stephane, 2017-09-27:
The API has changed between http://libmodbus.org/docs/v3.0.6/modbus_set_response_timeout.html
http://libmodbus.org/docs/v3.1.4/modbus_set_response_timeout.html

Could you really check you use the latest version by printing it?
You could find an example here:
https://github.com/stephane/libmodbus/blob/master/tests/version.c

IlhanSeven, 2017-09-27:
Hi,

I could not compile it I am sorry. This is the output:

`versionn.c:6:85: error: 'LIBMODBUS_MAJOR_VERSION' undeclared (first use in this function)
versionn.c:6:85: note: each undeclared identifier is reported only once for each function it appears in
versionn.c:6:85: error: 'LIBMODBUS_MINOR_VERSION' undeclared (first use in this function)
versionn.c:6:85: error: 'LIBMODBUS_MICRO_VERSION' undeclared (first use in this function)`

By the way I installed the library with opkg by 
opkg install libmodbus5

Thanks
Ilhan

zzeroo, 2017-09-27:
@IlhanSeven I bet this is libmodbus version 3.0.6. You have to use this API then: http://libmodbus.org/docs/v3.0.6/modbus_set_response_timeout.html

Because: https://reposcope.com/package/libmodbus5

IlhanSeven, 2017-09-27:
Thanks guys, I figured my version is 3.0.3. It was a version collision. Issue now can be closed. Although my CRC error, timeout issues persist.

Cheers.

stephane, 2017-09-28:
Could you close it, please?

closed 2017-09-28 by IlhanSeven