stephane/libmodbus #406: Add a new function modbus_send_raw_msg

musedavid, 2017-10-30:
Sometimes, sending a raw message without any changing is needed for debug purpose or modbus and other device that not useing modbus connected with the same serial rs485.
I hope a new function list bellow add to api.
modebus.h
MODBUS_API int modbus_send_raw_msg(modbus_t *ctx, uint8_t *msg, int msg_length);
modbus.c
int modbus_send_raw_msg(modbus_t *ctx, uint8_t *msg, int msg_length)
{
    int rc;
    int i;

    if (ctx->debug) {
        for (i = 0; i < msg_length; i++)
            printf("[%.2X]", msg[i]);
        printf("\n");
    }

    /* In recovery mode, the write command will be issued until to be
       successful! Disabled by default. */
    do {
        rc = ctx->backend->send(ctx, msg, msg_length);
        if (rc == -1) {
            _error_print(ctx, NULL);
            if (ctx->error_recovery & MODBUS_ERROR_RECOVERY_LINK) {
                int saved_errno = errno;

                if ((errno == EBADF || errno == ECONNRESET || errno == EPIPE)) {
                    modbus_close(ctx);
                    _sleep_response_timeout(ctx);
                    modbus_connect(ctx);
                } else {
                    _sleep_response_timeout(ctx);
                    modbus_flush(ctx);
                }
                errno = saved_errno;
            }
        }
    } while ((ctx->error_recovery & MODBUS_ERROR_RECOVERY_LINK) &&
             rc == -1);

    if (rc > 0 && rc != msg_length) {
        errno = EMBBADDATA;
        return -1;
    }

    return rc;
}

karlp, 2017-10-30:
If you want raw raw, why not just grab the fd from the object and call write() on it yourself?

musedavid, 2017-10-31:
There isn't an easy way to grab the fd from the object in libmodbus.
I need include  "modbus-private.h", and convert modbus_t to _modbus, then get backend_t.
As for me that isn't a the perfect solution。

pboettch, 2017-10-31:
No need to include the private.h, there is a public API to get the current file-descriptor from modbus-context.
```C
int modbus_get_socket(modbus_t *ctx);
````

musedavid, 2017-10-31:
The way above will not work, when using rs485.
I must hack deep and get more  private variables and functions  for the flow control (RTS). 
Codes looks like bellow, 
modbus_rtu_t *ctx_rtu = ctx->backend_data;
    if (ctx_rtu->rts != MODBUS_RTU_RTS_NONE) {
        ssize_t size;

        if (ctx->debug) {
            fprintf(stderr, "Sending request using RTS signal\n");
        }

        ctx_rtu->set_rts(ctx, ctx_rtu->rts == MODBUS_RTU_RTS_UP);
        usleep(ctx_rtu->rts_delay);

        size = write(ctx->s, req, req_length);

        usleep(ctx_rtu->onebyte_time * req_length + ctx_rtu->rts_delay);
        ctx_rtu->set_rts(ctx, ctx_rtu->rts != MODBUS_RTU_RTS_UP);

        return size;

pboettch, 2017-11-01:
(For readability could you edit your posts and put  \```C and \``` around your code? You'll get syntax highlighting for free).

musedavid, 2017-11-01:
Cool , I got it. A very useful skill