stephane/libmodbus #409: How can we help - new releases?

pboettch, 2017-11-01:
Hi @stephane, 

except reviewing PRs, replying to issues and waiting, how can we support you to get more releases done, creating development/next-version-branches and so on?

There is a whole bunch of PRs which could be integrated, there is quite a number of forks which a worth taking a look at.

I'm here since one year and can spend some time on the project. There are others who are appearing regularly as well - and I'm sure - willing to help.

stephane, 2017-11-27:
I intend to dedicate time on the project new week to define a plan.

agutenkunst, 2018-05-09:
Any updates on a new release? Would really appreciate it :+1: !

stephane, 2022-10-03:
The situation is better these days.
More funding more releases...

closed 2022-10-03 by stephane