stephane/libmodbus #412: Does not seem to support multi-threading

luocheng610, 2017-12-07:
Please read the following carefully before submitting this new issue.

 - Please ensure, that you are really reporting a bug. When in doubt,
   post a message on https://groups.google.com/forum/#!forum/libmodbus
   or send an email to libmodbus@googlegroups.com

 - Please do not open issues to ask questions about using libmodbus.
   Use the mailing list for this as there are many more people reading
   that list, who could help you.

 - When using libmodbus from a distribution (Debian, Fedora...), please
   report the bug first in the bug tracker of the distribution. The
   reason for doing so is that the package maintainer should have a chance
   to look at the issue first as it might be a packaging error. If/when
   the package maintainer comes to the conclusion that is really an upstream
   bug, then he/she will usually report it here by himself/herself.
   This is because he/she is interested in staying in the notification chain
   to decide about a backport as soon as a bugfix is available.
   Otherwise you (distribution user) will be asked to do so explicitely.

When you get here and you are still convinced that you want report a bug:

 - *Use a clear and decriptive title* for the issue to identify

 - *Which version of libmodbus are you using?* you can obtain this information
   from your package manager or by running `pkg-config --modversion libmodbus`.
   You can provide the sha1 of the commit if you have fetched the code with `git`.

 - *Which operating system are you using?*

 - *Describe the exact steps which reproduce the problem* in as many details as
   possible. For example, the software/equipement which runs the Modbus server, how
   the clients are connected (TCP, RTU, ASCII) and the source code you are using.

 - *Enable the debug mode*, libmodbus provides a function to display the content
   of the Modbus messages and it's very convenient to analyze issues
   (http://libmodbus.org/docs/latest/modbus_set_debug.html).

Good bug reports provide right and quick fixes!

Finally, thank you very much for using libmodbus and taking the time to
file a good bug report. Doing so signals your respect for the developers.

The following template helps you to address the points above. Please delete
everything up to and including the following line which starts with ---.

---

## libmodbus version

  <libmodbus version or sha1 of the latest commit>

## OS and/or distribution

  <e.g. Windows, Linux... version>

## Environment

  <e.g. CPU architecture, 32 vs. 64 bit...>

## Description

  <...>

## Expected behaviour

  <...>

## Actual behaviour

  <...>

## Steps to reproduce the behavior (commands or source code)

  <...>

## libmodbus output with debug mode enabled

  <...>

karlp, 2017-12-07:
aiui, this should work just fine, as long as you don't try and access the modbus context concurrently.  so, your mutex isn't mutexing....

There's been some issues filed before with other advice, and also on the mailing list, that might be of interest

gsaviane, 2018-04-06:
I've been working for developing a Modbus/TCP gateway. The gateway plays a virtual slave device role and many concurrent masters can connect to it for sending Modbus messages that are forwarded to other networks. It seems not possible to manage concurrent requests in parallel since the context is unique and for processing each request its socket handle must be set in the context with modbus_set_socket(context, handle).
That means that each thread managing a request must enter a critical section in order to avoid the context being shared with other threaded requests. 
 It seems to be a design limit. One possible "dirty" solution is to make a copy of the context for each request, but I doubt it can work as the context is an opaque object that normally the user has no access to its structural declaration.

Anybody can confirm/deny my thesis?

Thank you

pboettch, 2018-05-21:
I'm doing exactly what you describe (for TCP and RTU) within a single thread using one context. It works like a charm (with the modifications done on my callback-based server/slave implementation). One might think that you need multiple threads to handle multiple clients, but you don't actually, `poll()` works fine.

stephane, 2019-07-28:
I don't know what do you propose to improve the situation. To discuss further (if required) on the mailing list.

closed 2019-07-28 by stephane