stephane/libmodbus #415: Dead connections

Tangerino, 2017-12-21:
How do we detect dead connections on a modbus server?
How to avoid clients to connect and not send data or became unresponsive? Server may drop the connections from lazy clients.
Thanks in advance

watsocd, 2017-12-21:
From the documentation: http://libmodbus.org/docs/v3.0.6/modbus_read_registers.html or http://libmodbus.org/docs/v3.1.4/modbus_read_registers.html 

 

The modbus_read_registers() function shall return the number of read registers if successful. Otherwise it shall return -1 and set errno.

 

If not connected or read fails, try to reconnect and read again. 

 

Remote devices/servers that fail often can drastically affect the total efficiency of the entire MODBUS network. Especially with RTU/serial networks. It is just the nature of a master/slave type of network.

Tangerino, 2017-12-23:
Never mind, the multi client example does not handle timeouts if we connect AND do not send anything. That will allocate all available sockets. This solution is outside the library context. Anyway be aware that DOS attack is easy to reproduce.
It would be nice to have a void *user_data in the context so we can associate a timer for instance for each connection and close an idle socket. I would change the SELECT call to accept a time out argument.
Thank you.

fkgit, 2018-01-01:
->main
The client connection from 192.168.1.4 is accepted
Connecting to 192.168.1.100:102
Waiting for a indication...

Excuse me, what's the reason for this?

oazradSPOC, 2020-11-20:
have you found the answer for this. I'm writing a server application and it works well. but if my client gets interrupted my slave program will hang and will not recover even if I cycle power on my client.

stephane, 2024-10-22:
I deployed various Modbus servers powered by libmodbus and it's easy to close/open dead connections on error. The lib provides helpers (https://libmodbus.org/reference/modbus_set_error_recovery/)

closed 2024-10-22 by stephane