stephane/libmodbus #417: Does not compile unter Ubuntu, problem with configure/autogen?

SebastianBT, 2018-01-05:
Hello,

I tried to compile Libmodbus(cloned the Repository) under Ubuntu, but it dies not seem to work.

**Version 3.0.6:**
_sudo ./configure …. Works
sudo make …. Works_

**Version 3.1.4:**
_sudo ./configure …. Works
sudo make …. Works_

**Git Clone:**

_sudo: ./configure: command not found_

_sudo ./autogen.sh
configure.ac:33: installing 'build-aux/compile'
configure.ac:56: installing 'build-aux/config.guess'
configure.ac:56: installing 'build-aux/config.sub'
configure.ac:32: installing 'build-aux/install-sh'
configure.ac:32: installing 'build-aux/missing'
src/Makefile.am:2: error: Libtool library used but 'LIBTOOL' is undefined
src/Makefile.am:2:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
src/Makefile.am:2:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
src/Makefile.am:2:   If 'LT_INIT' is in 'configure.ac', make sure
src/Makefile.am:2:   its definition is in aclocal's search path.
src/Makefile.am: installing 'build-aux/depcomp'
parallel-tests: installing 'build-aux/test-driver'
autoreconf: automake failed with exit status: 1_

_--------------------------_
_Running autoreconf failed._
_--------------------------_

_sudo ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for gcc option to accept ISO C99... none needed
checking for gcc option to accept ISO Standard C... (cached) none needed
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files... no
checking whether make supports nested variables... (cached) yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
./configure: line 5320: syntax error near unexpected token `disable-static'
./configure: line 5320: `LT_INIT(disable-static win32-dll pic-only)'_

Thank you very much,

Best Regards,

Sebastian

mhei, 2018-01-05:
Quote from README.md:

> You will only need to install automake, autoconf, libtool and a C compiler (gcc or clang)
> to compile the library and asciidoc and xmlto to generate the documentation (optional).

Looks like you missing the libtool part. Use ` sudo apt-get install libtool` to install it.

SebastianBT, 2018-01-05:
Thank you, problem solved, no Bug.

closed 2018-01-05 by SebastianBT

zxyAcmen, 2020-04-26:
All the necessary tools are installed, but this problem still occurs,
./configure: line 5320: syntax error near unexpected token `disable-static'
./configure: line 5320: `LT_INIT(disable-static win32-dll pic-only)'
help me

zxyAcmen, 2020-04-26:
configure: WARNING: unrecognized options: --enable-static
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for arm-linux-strip... no
checking for strip... strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking for style of include used by make... GNU
checking for arm-linux-gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for gcc option to accept ISO C99... none needed
checking for gcc option to accept ISO Standard C... (cached) none needed
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files... no
checking whether make supports nested variables... (cached) yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... arm-unknown-linux-gnu
./configure: line 5320: syntax error near unexpected token `disable-static'
./configure: line 5320: `LT_INIT(disable-static win32-dll pic-only)'

zxyAcmen, 2020-04-26:
The problem is solved. Download libtool and follow the steps again