stephane/libmodbus #418: Cannot set rs485 mode : ubuntu on phoenix industrial pc with hardware rs485

yarula, 2018-01-15:
I have an industrial Phoenix PC by hand with rs485 hardware support.
Enable this mod in bios for ttyS1.
This PC wired to kwh meter by appropriate cable.

Before I was successfull in reading data from the meter using rs485 to rs232 converter wihtout setting rs485 mode on by libmodbus call.

Now I want to drop converter and use native rs485 .
```

...

modbus_t *ctx;
ctx = modbus_new_rtu(_device_path.c_str(), _bod, _parity[0], _data_bit, _stop_bit);
if (ctx == NULL) {
        modbus_free(ctx);
          throw runtime_error("Unable to create the libmodbus context");
}

modbus_set_slave(ctx, 1);
if (modbus_connect(ctx) == -1) {
        string msg("Connection failed: "); msg += modbus_strerror(errno);
        modbus_free(ctx);
        throw runtime_error(msg);
}

if (modbus_rtu_set_serial_mode(ctx, MODBUS_RTU_RS485) < 0 ) {
          int mode = modbus_rtu_get_serial_mode(ctx);
          modbus_free(ctx);
          string msg("Cannot set rs485 mode "); msg += modbus_strerror(errno);
          throw runtime_error(msg);
}
...
```

so I see 

> Cannot set rs485 mode Inappropriate ioctl for device

Need your advice :(

yarula, 2018-01-15:
if to call 
>modbus_rtu_get_serial_mode
it returns 
>MODBUS_RTU_RS232

Does it mean I should do something with driver ?

karlp, 2018-01-15:
the serial mode setting is for drivers that support switching from userspace.

If you have a bios setting that just turns it on automatically, there's nothing you need to do in your application, it's transparent.  Just don't try and set the mode.

yarula, 2018-01-15:
Thanks for quick reply!
It clears much.

After removing setting the mode code I get "connection timeout" or "too many data" errors

I'l thy to dig it...

jcwren, 2018-01-15:
I recently modified the driver for the Raspberry PI 3 to drive the RTS line
for RS-485 support. If the driver for the UART on your PC doesn't support
that, it's likely not too difficult to add. Fundamentally, the function is
the same. When you transmit you drive the RTS line to the appropriate
state, and when the transmitter buffer is empty, you flip it back. There's
a handful of drivers that already support this, and you could use them for
reference.

--jc

Save a life; adopt a shelter animal.

yarula, 2018-01-15:
Where I could get this driver to test?

yarula, 2018-01-15:
now I'm struggling with 
ERROR Connection timed out: select

yarula, 2018-01-15:
>If you have a bios setting that just turns it on automatically, there's nothing you need to do in your >application, it's transparent. Just don't try and set the mode.

But I checked this one: I called get_rtu_serial_mode and received MODBUS_RTU_RS232

mhei, 2018-01-16:
As it was already pointed out, this seems to be expected behaviour: from the point of view of your operating system, the port is still a simple UART. It does not see whether there is a RS232 or RS485 physical line driver chip enabled by the BIOS before the signal is routed to the plug. So your kernel still assumes this is RS232 and this is correct so.
BTW: there should be some documentation whether you need RTS for driver enable, or whether this is also transparent.

If you run into timeout, then double-check that there is really some physical output at the port, e.g. using a second PC and your existing RS232->RS485 converter. Use HTerm or similar as terminal program to see whats going on...

yarula, 2018-01-17:
I have 2-wire rs485.
First I double-checked the wire... The pin schema was suspected : 
5-pin: gnd ( ok )
2-pin: A
3-pin: B
I suppose correct schema is :  

5-pin: gnd ( ok )
7-pin: A
3-pin: B

I wired it, then connect PC and slave device, then sun the interceptty sinffer an my code.

Still the same: connection time out,  and interceptty show outgoing ( from PC )  bytes  (read_register_request) and no incoming .

http://opengear.com/images/sd4001-pinout.png

yarula, 2018-01-17:
update:  I ringed serial pc pins and found that 5,2,3-schema corrsponds to current pins voltage. So it's strange anough but it is ok.
So the question is the same , why I'm not still able to receive data from the slave.

AAG-ch, 2018-01-17:
Hello, I have the same problem. I'm using [ITX-M51-D826](http://www.tostarcn.com/prod_view.aspx?TypeId=73&Id=306) with build in RS485. I have enabled RS485 with a jumper. I am connecting it with another PC using USB converter. On another PC I'm using modbus slave server, on ITX this code:
```
int main(void) {
	modbus_t *ctx;
	uint16_t tab_reg[64];
	int rc;
	int i;
	ctx = modbus_new_rtu("/dev/ttyS4", 38400, 'N', 8, 1);
	if (ctx == NULL) {
	    fprintf(stderr, "Unable to create the libmodbus context\n");
	    return -1;
	}
	modbus_set_slave(ctx, 99);
//	modbus_rtu_set_serial_mode(ctx, MODBUS_RTU_RS485);
//	modbus_set_response_timeout(ctx, 3, 200000);
//	modbus_rtu_set_rts(ctx, MODBUS_RTU_RTS_UP);
//	modbus_rtu_set_rts_delay(ctx, 1000);
	modbus_set_debug(ctx, TRUE);
    	modbus_set_error_recovery(ctx,
                              MODBUS_ERROR_RECOVERY_LINK |
                              MODBUS_ERROR_RECOVERY_PROTOCOL);

	if (modbus_connect(ctx) == -1) {
	    fprintf(stderr, "Connection failed: %s\n", modbus_strerror(errno));
	    modbus_free(ctx);
	    return -1;
	}
	rc = modbus_read_registers(ctx, 1, 1, tab_reg);
	if (rc == -1) {
	    fprintf(stderr, "%s\n", modbus_strerror(errno));
	    return -1;
	}
	for (i=0; i < rc; i++) {
	    printf("reg[%d]=%d (0x%X)\n", i, tab_reg[i], tab_reg[i]);
	}
	modbus_close(ctx);
	modbus_free(ctx);
}
```
Output of the programm:
> Opening /dev/ttyS4 at 38400 bauds (N, 8, 1)
[63][03][00][01][00][01][DD][88]
Waiting for a confirmation...
ERROR Connection timed out: select
Bytes flushed (0)
Connection timed out

But in communication console on my PC i'm geting:
> RX: 63 03 00 01 00 01 DD 88
TX: 63 03 02 00 01 80 4C

It means PC generate response but ITX did't receive data from the slave. Is this a common problem?

yarula, 2018-01-24:
little update:
I prepared windows IoT as dual boot on a problem device.
Runned the same test program and received register data from the 1st try.

Returned back to linux ( I reinstalled it for pure experiment ) - no answer.

My conclusion is that: I have a problem with serial driver in linux.
ubuntu 16.04, kernel 4.4.0-87

Nuvoton communications port is a device manufacturer. 

So, I'm gonna dig driver question...

Any advice?

VishvasMirani, 2023-03-22:
> Hello, I have the same problem. I'm using [ITX-M51-D826](http://www.tostarcn.com/prod_view.aspx?TypeId=73&Id=306) with built in RS485. I have enabled RS485 with a DIP-switch. I am connecting it with another PC using USB converter. On another PC I'm using modbus slave server, on ITX this code:
> 
> ```
> int main(void) {
> 	modbus_t *ctx;
> 	uint16_t tab_reg[64];
> 	int rc;
> 	int i;
> 	ctx = modbus_new_rtu("/dev/ttyS4", 38400, 'N', 8, 1);
> 	if (ctx == NULL) {
> 	    fprintf(stderr, "Unable to create the libmodbus context\n");
> 	    return -1;
> 	}
> 	modbus_set_slave(ctx, 99);
> //	modbus_rtu_set_serial_mode(ctx, MODBUS_RTU_RS485);
> //	modbus_set_response_timeout(ctx, 3, 200000);
> //	modbus_rtu_set_rts(ctx, MODBUS_RTU_RTS_UP);
> //	modbus_rtu_set_rts_delay(ctx, 1000);
> 	modbus_set_debug(ctx, TRUE);
>     	modbus_set_error_recovery(ctx,
>                               MODBUS_ERROR_RECOVERY_LINK |
>                               MODBUS_ERROR_RECOVERY_PROTOCOL);
> 
> 	if (modbus_connect(ctx) == -1) {
> 	    fprintf(stderr, "Connection failed: %s\n", modbus_strerror(errno));
> 	    modbus_free(ctx);
> 	    return -1;
> 	}
> 	rc = modbus_read_registers(ctx, 1, 1, tab_reg);
> 	if (rc == -1) {
> 	    fprintf(stderr, "%s\n", modbus_strerror(errno));
> 	    return -1;
> 	}
> 	for (i=0; i < rc; i++) {
> 	    printf("reg[%d]=%d (0x%X)\n", i, tab_reg[i], tab_reg[i]);
> 	}
> 	modbus_close(ctx);
> 	modbus_free(ctx);
> }
> ```
> 
> Output of the programm:
> 
> > Opening /dev/ttyS4 at 38400 bauds (N, 8, 1)
> > [63][03][00][01][00][01][DD][88]
> > Waiting for a confirmation...
> > ERROR Connection timed out: select
> > Bytes flushed (0)
> > Connection timed out
> 
> But in communication console on my PC i'm geting:
> 
> > RX: 63 03 00 01 00 01 DD 88
> > TX: 63 03 02 00 01 80 4C
> 
> It means PC gets request and generate response, but ITX did't receive data from the slave. Is this a common problem?

 i am trying to same and i am also suffering from the same problem.

stephane, 2024-10-22:
The discussion seems related to hardware support issues.