stephane/libmodbus #422: config.h no such file or directory

Yoshimiga, 2018-02-11:
I am trying to compile the source code but i keep getting this error
master@Master-Virtual:~/Downloads/libmodbus-master/src$ gcc *.c
modbus.c:21:20: fatal error: config.h: No such file or directory
compilation terminated.
modbus-data.c:24:20: fatal error: config.h: No such file or directory
compilation terminated.
In file included from modbus-rtu.c:17:0:
modbus-private.h:19:20: fatal error: config.h: No such file or directory
compilation terminated.
modbus-tcp.c:25:37: fatal error: config.h: No such file or directory
compilation terminated.

I checked to make sure every source file had #include <config.h> file i tried moving config.h into the src directory but i still get this error. I know that i must somehow manually set a path for the header file in order for the compiler to know where to look for it while compiling but i have no idea how to do that. I have tried numerous attempts yet the gcc is not finding the header file. I have successfully compiled the contents  in the "tests" folder without any issues and were able to run the executables. I have read the README file over 100 times and followed its instructions, however its not clear when it comes to the issue that i am experiencing. Would anyone please help, and thank you.

pboettch, 2018-02-11:
Please read the installation section of the README-file:

https://github.com/stephane/libmodbus#installation

Yoshimiga, 2018-02-12:
@pboettch i have read it many times... as i mentioned in my initial post.

pboettch, 2018-02-12:
Then what makes you try to compile it using `gcc *.c` ?

If you want to use this library from source, you'll need to build it (with `configure` and `make`), then install it.

After installation you can build and link your application with it.

Yoshimiga, 2018-02-13:
@pboettch Mr. Patrick i initially ran the autogen.sh file to generate the configure script. Then ran the ./configure, and got a confirmation that everything was installed. I used sudo make install to install the build, everything installed. (Libraries have been installed in:
   /usr/local/lib) I also ran the config.status and made sure everything was up to date. Next i ran sudo ldconfig for linkage. I used an example from tests directory README file gcc random-test-server.c -o random-test-server `pkg-config --libs --cflags libmodbus` and it compiled. In the "tests" directory folder everything got compiled and i got executables from all the .c source codes inside the tests directory. I was able to run them without any issues.
Similarly i tried using the same gcc modbus.c -o modbus `pkg-config --libs --cflags libmodbus` in the src directory, but what ever i try to compile in it just leaves me with the error that in modbus.c:21:20: fatal error: config.h: No such file or directory compilation terminated. 
So at this point im not sure why the compiler is not picking up the config.h file if it is included inside the source code #include <config.h> I manually tried placing the config.h header file into the src directory but got the same error. 
From my own research i saw that i must specify path directory to the config.h file in order for the gcc compiler to pick it up, because as of now it seems it is not locating this config.h file even though it is in the proper directory. Im not sure how to set this path. Am i right about this issue? Did other users of this library came up with a similar issue? I have checked the google discussion thread about stephane libmodbus library and also read all the issues on github and have not found anyone experiencing similar issue? Must i try to compile .c source codes in any given order if i want to run a tcp based communication? (I know i must swap ip addresses of two different machines inside source code for Client and Server to communicate. I have studied Modbus protocol carefully and have a good idea how socket programming works.)
Would you please point me in the right direction because i have been stuck on this issue for a while now and i want set up communication between Client and the Server so i can continue with my project. Any help would be greatly appreciated. Sorry for the long read but i wanted to include all details. Thank you for responding and thank you for you time.

stephane, 2018-06-05:
Please use the libmodbus mailing list for help.

closed 2018-06-05 by stephane

mpm756769, 2019-11-20:
I am getting same error as Yoshimiga.

Provide some solution for same if possible.Thank you.