stephane/libmodbus #423: modbus double value conversion function

alibasel, 2018-03-06:
I started to use siemens sentron pac3200 energy analyser with this library.

For the address 801 a double value returns like below:
<29><E8><00><00><00><0B><FF><03><08><41><E5><61><A9><ED><DE><6E><39>
And this the value part 8 bytes - 4 words: <41><E5><61><A9><ED><DE><6E><39>

I need a similar conversion function like for the float values but for double values
float modbus_get_float_dcba(const uint16_t *src)

double modbus_get_double_dcba(const uint16_t *src)

Can anyone convert this function to double?

/* Get a float from 4 bytes (Modbus) in inversed format (DCBA) */
float modbus_get_float_dcba(const uint16_t *src)
{
    float f;
    uint32_t i;

    i = ntohl(bswap_32((((uint32_t)src[0]) << 16) + src[1]));
    memcpy(&f, &i, sizeof(float));

    return f;
}

pboettch, 2018-03-07:
> For the address 801 a double value returns like below: `<29><00><00><00><0B><03><08><41><61><6E><39>`
> And this the value part 8 bytes - 4 words: `<41><61><6E><39>`

Here you mix up some thing: `<41>` is actually one byte, so `<41><61><6E><39>` are 4 bytes and not 4 words (if one word is 16 bit). 

If you want to read a double-value (which is 8 bytes) you need to read 4 16-bit words (or registers)

```C
    uint16_t src[4];
    int ret = modbus_read_registers(ctx, 801, 4, src);
```
Then to convert it to double, well that depends on your platform and your compiler. I use `uint64_t` here, this may or may not work on your host (to be sure your host needs to be a 64 bits platform).

```C
#include <stdint.h>
#include <endian.h>

double modbus_get_double_from_big(const uint16_t *src)
{
	uint64_t v;
	v  = (uint64_t) src[0] << 48;
	v |= (uint64_t) src[1] << 32;
	v |= (uint64_t) src[2] << 16;
	v |= (uint64_t) src[3] << 0;
#if __BYTE_ORDER == __LITTLE_ENDIAN
	v = __bswap_64(v);
#endif
	return *(double *) &v;
}
```

alibasel, 2018-03-07:
Pboettech,

Thank you for your response. You are rigth, it seems that i missed the last parts of the values while copying.
I tried your function, but the resulting double value does not seem correct. I'm using a gentoo 64 bit pc and gcc 5.4.0.
Here are some values that were supplied and their results.
Could you please check them:

tcpmodbus response: <00><70><00><00><00><0B><FF><03><08><41><D2><69><4E><DA><45><22><9A>
Address=801,  Supplied value: 0x41d2-0x694e-0xda45-0x229a- Returned Double Value= -0.000000

tcpmodbus response: 
<00><78><00><00><00><0B><FF><03><08><41><D2><69><4E><DA><C2><90><46>
Address=801,  Supplied value: 0x41d2-0x694e-0xdac2-0x9046- Returned Double Value= 84989094928568993035691348721664.000000

alibasel, 2018-03-07:
Sorry, again a copy paste mistake:

tcpmodbus response: <00><70><00><00><00><0B><03><08><41><D2><69><4E><45><22><9A>
 Address=801, Supplied value: 0x41d2-0x694e-0xda45-0x229a- Returned Double Value= -0.000000

tcpmodbus response:
 <00><78><00><00><00><0B><03><08><41><D2><69><4E><DA><C2><90><46>
 Address=801, Supplied value: 0x41d2-0x694e-0xdac2-0x9046- Returned Double Value= 84989094928568993035691348721664.000000

alibasel, 2018-03-07:
Sorry, but I don't know why, this editor cuts some of the values in tcpmodbus response lines when I publish my message. Please consider the supplied value sections, they are correct.

pboettch, 2018-03-07:
(Please read https://guides.github.com/features/mastering-markdown/ to create super-duper formatted comments on github - and you can edit posts).

What is the expected value? Do you have a spec for this device describing the registers?

pboettch, 2018-03-07:
OK, the swap is not need.

I get 

  1235565417.080237 
  1235565419.040056

from your values with this function

```C
double modbus_get_double(const uint16_t *src)
{
	uint64_t v;
	v  = (uint64_t) src[0] << 48;
	v |= (uint64_t) src[1] << 32;
	v |= (uint64_t) src[2] << 16;
	v |= (uint64_t) src[3] << 0;
	return *(double *) &v;
}
```

alibasel, 2018-03-07:
Thank you pboettech, that's worked, you are great!
I think you should submit this function for others use also.
Regards,
Ali

pboettch, 2018-03-07:
Not sure whether this helper-function has its place in the library, it is more the device which defines this way of interpreting values. 

Would you mind closing this issue if you consider it solved for you? Thanks.

alibasel, 2018-03-07:
Patrick,
float modbus_get_float_dcba(const uint16_t *src) and similar conversion functions  are defined in the src/modbus-data.c file. So your double conversion function also can have  its place there.
Thank you again for your support, I'm closing this issue.

closed 2018-03-07 by alibasel