stephane/libmodbus #424: add a sniffing function for RTU serial line modbus

gitaeuber, 2018-03-10:
Hi everybody,

for the network modbus there is the wireshark tool for sniffing. But there is none for the serial line (RS485) for the (Linux) console afaik.

Please add some functions to sniff the protokoll on serial lines.
Maybe some functions like:
modbus_sniff(ctx)
that could be called by
modbus_sniff_and_parse(ctx)

Thanks a lot and best regards.

pboettch, 2018-03-10:
Patches are welcome. Do you think you could contribute something?

gitaeuber, 2018-03-11:
> Patches are welcome. Do you think you could contribute something?

I don't think I'm able to contribute useful code. I'm not really a coder. Sorry.

karlp, 2018-03-11:
Is a patch for this even actually welcome? it seems way out of scope

pboettch, 2018-03-11:
Why not adding debug and analyze code to the library, for example in a tools-dir? We could still debate on a patch once it's there. IMHO.

gitaeuber, 2018-03-16:
Hi there!

This is a try to introduce a function for sniffing on a serial line. It isn't complete yet.
The attached diff is a patch against
`#define LIBMODBUS_VERSION        3.1.4
`

What do you think about this?

I get compiling errors I can't resolve. Could you help me?

```
.libs/modbus-data.o:(.data.rel.local+0x0): multiple definition of `ucl_msg_type'
.libs/modbus.o:(.data.rel.local+0x0): first defined here
.libs/modbus-rtu.o:(.data.rel.local+0x0): multiple definition of `ucl_msg_type'
.libs/modbus.o:/tmp/libmodbus/src/modbus.c:45: first defined here
.libs/modbus-tcp.o:(.data.rel.local+0x0): multiple definition of `ucl_msg_type'
.libs/modbus.o:/tmp/libmodbus/src/modbus.c:45: first defined here
collect2: error: ld returned 1 exit status
Makefile:447: recipe for target 'libmodbus.la' failed
make[2]: *** [libmodbus.la] Error 1
Makefile:500: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
Makefile:386: recipe for target 'all' failed
make: *** [all] Error 2

```

Thanks and best regards
Lars
[sniff_serial.diff.gz](https://github.com/stephane/libmodbus/files/1818364/sniff_serial.diff.gz)

_PS: Is there an open mailinglist somewhere? I don't have a google account and don't want one._

pboettch, 2018-03-16:
Please use Github's Pull-request feature for code-reviews. (Clone this repo onto your account, push to it and then "Create Pull Request")

pboettch, 2018-03-16:
Just mention in your PR that it doesn't work and that you just want a review. People are generally nice here because all useful contribution efforts are seen as a great thing.

gitaeuber, 2018-03-16:
I managed to make a pull request. I had to fork the repo to my account first.
Thanks

pboettch, 2018-03-16:
That's what I wanted to say with clone to your account. Fork is the right word.

embcla, 2021-01-20:
Hello 

Has this functionality been made available in the general library?
I've got a use case where I need to monitor traffic between two endpoints without putting additional traffic on the bus

gresolio, 2023-02-22:
QModMaster uses two custom callbacks to implement Bus Monitor functionality:

* [modbus.c --> send_msg() --> busMonitorRawRequestData()](https://sourceforge.net/p/qmodmaster/code/ci/default/tree/3rdparty/libmodbus/modbus.c#l179)
```C
//***Not part of libmodbus - added for QModMaster***//
busMonitorRawRequestData(msg, msg_length);
```

* [modbus.c --> _modbus_receive_msg() --> busMonitorRawResponseData()](https://sourceforge.net/p/qmodmaster/code/ci/default/tree/3rdparty/libmodbus/modbus.c#l474)
```C
//***Not part of libmodbus - added for QModMaster***//
busMonitorRawResponseData(msg, msg_length);
```

Do you think it makes sense to add registration of such callbacks? For example, something like this:
```C
typedef void modbus_raw_monitor_handler(uint8_t* data, int len);

int modbus_add_raw_request_monitor(modbus_raw_monitor_handler* handler);
int modbus_add_raw_response_monitor(modbus_raw_monitor_handler* handler);
```