stephane/libmodbus #429: Modbus Rtu Sedona cihazında veri aktarımı sırasında sedona sistemini durduruyor

coskunyildirim, 2018-03-31:
Sedona Framework also adds libmodbus as a native file,
I've created.
However, Modbus Rtu is in a state of pause in Sedona during communication.
This pause continues until the data transfer is over. Sedona is working normally after the transfer is finished.
Sedona Framework works in raspberry pi3.
I think this problem originated in the libmodbus library.
But I do not know how to find a solution.
Dear friends, I ask for your help in this regard.
Best regards
Coşkun YILDIRIM

karlp, 2018-04-03:
does sedona do things like pretend to be a posix system, but actually fake out poll/select?

coskunyildirim, 2018-04-03:
It works exactly the same way. There is a sequence of each process in sedona.
There is a conflict, but I do not know how to solve it.

stephane, 2018-06-05:
Not a bug report. Sorry.

closed 2018-06-05 by stephane