stephane/libmodbus #430: modbus_tcp_accept freezes in debugger [cpp]

hanusek, 2018-03-31:
## libmodbus version
libmodbus v3.1.4 

## OS and/or distribution
Ubuntu Linux 16.04.3 LTS

## Environment
x86_64
gcc/g++ 4.9
and 
gcc/g++ 5.4.1

## Description
Attempt to use the libmodbus library in **C++** code.

## Actual behaviour
The **modbus_tcp_accept** function freezes, stands still.
The debugger does not go to the next breakpoint.

## Steps to reproduce the behavior (commands or source code)
```
void ExampleProgram::start()
{
    std::cout << "ExampleProgram::start()" << std::endl;

    _ctx = modbus_new_tcp("127.0.0.1", 2055);
    if(_ctx == nullptr)
    {
        std::cout << "_ctx is nullptr" << std::endl;
        return;
    }
    _s      = modbus_tcp_listen(_ctx, 1);
    int res = -1;
    res     = modbus_tcp_accept(_ctx, &_s);
    if(res == -1)
    {
        return;
    }
    _mbMapping = modbus_mapping_new(MODBUS_MAX_READ_BITS, 0, MODBUS_MAX_READ_REGISTERS, 0);
    if(_mbMapping == nullptr)
    {
        std::cout << "_mbMapping is nullptr" << std::endl;
        // fprintf(stderr, "Failed to allocate the mapping: %s\n", modbus_strerror(errno));
        modbus_free(_ctx);
        return;
    }
    std::cout << "ExampleProgram::start() end" << std::endl;
}
```

## libmodbus output with debug mode enabled
```
Debugging starts
Opening : /home/mhanusek/work/VersionControl/OpenPAC/build-pac-Desktop_Qt_5_9_1_GCC_64bit-Debug/ExampleProgram/libexample-program.so
ExampleProgram::start()
```

pboettch, 2018-04-01:
Did you try to connect a client to your server?  The expected behavior is that `modbus_tcp_accept(_ctx, &_s);` will block until a client is connecting.

If you want something not blocking your main-thread try to use `std::thread` and `poll()` (as I describe [here](https://github.com/stephane/libmodbus/issues/173#issuecomment-339013015) as an example).

closed 2018-04-05 by hanusek

sritam2, 2018-10-08:
@pboettch: I agree to your answer as my application also behaved the same. In my case there is no pending connection initially as no client is initiating a connection at the start. But server is first trying to initialize all its components and 1 of the components is modbus-tcp. But the call to "modbus_tcp_accept()"is becoming blocking. Is there any alternative way to make the call non-blocking or do I need to go for a multi-threading approach. Is there any signal or event notification mechanism.
Looking forward to your valuable advise.

Thanks and Regards,
Sritam Paltasingh.

pboettch, 2018-10-08:
@sritam2  The question you should ask yourself is, how much time is passing by between the "initialization of all its components" and the first client connection you want to handle and is this duration really critical to your system's behavior?

In general, in embedded software architectures you should avoid mixing up too many components in one application process (even with threads). This kind of architecture quickly leads to monster-processes which if they are crashing, nothing works anymore.

Think IPC or RPC.

pboettch, 2018-10-08:
Oh, I get it, it's only about the blocking aspect of `accept()`. 

Well, you can use the server-socket with `poll()` or `select()` to check whether there is activity on the socket. This, together with a timeout give you a kind of non-blocking behavior. You can also check whether the socket-option `O_NONBLOCK` works on server-sockets.

sritam2, 2018-10-08:
@pboettch: I am sorry for my late reply. I did investigation. if I change my server socket flag settings from O_BLOCK to O_NONBLOCK then my call to "modbus_tcp_accept()", which will eventually call accept() or accept4(), will fail if no client connection is there(which is my use case). Again I need to manually call modbus_tcp_accept() till it succeeds. So, best is to call poll() or select(). I will do further investigation and tell you the results.

Thanks and Regards,
Sritam Paltasingh.