stephane/libmodbus #448: TCP/IP connection from modbus master, doesn't work

LucaPolese, 2018-09-07:
Hi, I've been programming a Server modbus Slave that allows any Client slave to connect to it and read or write both coils and registers.
This is the second function of a bigger firmware to put in a device that can (on th basis of the JSON configuration) be a Gateway using OPCUA and Modbus protocols.

At this point, my code works and compiles, but, when I try to connect my ModScan, it freezes and doesn't go on.
Next you can find my ModBus Server code.

PS: I noticed that it stops on modbus_tcp_accept (); function.
https://hastebin.com/ajujiyeric.cpp

I attach what you can find on my WireShark
[image]

timmi-on-rails, 2018-10-05:
What are the actual values (ip, port) used for `ctx = modbus_new_tcp(slav.ip, slav.port);`?
Can you check the return code of `server_int = modbus_tcp_listen(ctx, 1);`

If `modbus_tcp_accept` is blocking it is usually waiting for an incoming connection.

stephane, 2019-07-29:
No answer. Closed

closed 2019-07-29 by stephane