stephane/libmodbus #451: Connection timed out and invalid data

bkormann, 2018-10-02:
Hi,

I am using libmodbus on a 64 Bit Debian Linux for a modbus tcp connection to a Siemens PAC 3200.
Reading of holding registers works basically. I get valid data from Siemens PAC. 
But if I do a sleep of 1 second and try to read registers again I receive timeouts and invalid data responses on a regular basis. It doesn't make any difference if I read 6 subsequent registers as a hole or read them register per register. In all cases I don't close the modbus connection. 
When looking at network traffic it turns out that the modbus server doesn't respond at a certain point in time anymore (see image attached).
Here is the code that I use:

modbus_t *mb;
uint16_t tab_reg[32];
int rc;
mb = modbus_new_tcp("10.203.159.85", 502);
if (modbus_connect(mb) == -1) 
{
  fprintf(stderr, "Connection failed:\n");
  modbus_free(mb);
   return -1;
}
while(1)
{
    int tags[10] = {1, 3, 5, 7, 9, 11, 13, 15, 17, 55};
    for(int i = 0; i < 10; i++)
    {
        rc = modbus_read_registers(mb, offset, 2, tab_reg);
        if ( rc == 2 )
        {
            float result = 0.0;
            int value = (tab_reg[0] << 16) + tab_reg[1];
            memcpy(&result, &value, sizeof(float));

            printf("Register: %d; Value: %f\n", tags[i], fValue);
        }
        else
        {
            fprintf(stderr, "Connection failed: %s\n", modbus_strerror(errno));
        }
    }
    sleep(1);
}

If I use another modbus tcp client (like ananas) there is no issue at all and I can even read registers at 250 ms cycles. 

Any ideas of how to solve the issue of losing connection or retrieving invalid data?

[image]

pboettch, 2018-10-02:
Hi bkormann, could you put your code-snippets inside a block of three backticks (\`\`\`)  to enable proper syntax highlight, it makes it easier to read.

pboettch, 2018-10-02:
What is the value of offset? Should it be 

    rc = modbus_read_registers(mb, tags[i], 2, tab_reg);

instead?

bkormann, 2018-10-02:
You are right. This was a typo. I update the code

pboettch, 2018-10-02:
Could you post the output of program running the code from above?

Is this code running in Linux or Windows? Could you run wireshark to see what happens on the network-level? And do the same when using "Ananas", with wireshark I mean.

This would allow us to see the difference between the two independently of any implementation.

bkormann, 2018-10-02:
This is a sample output:

> Register: 1; Value: 229.768738
Register: 3; Value: 229.503799
Register: 5; Value: 229.558502
Register: 7; Value: 397.765198
Connection failed: Connection timed out
Register: 9; Value: 0.000000
Connection failed: Connection timed out
Register: 11; Value: 0.000000
Connection failed: Invalid data
Register: 13; Value: 0.000000
Register: 15; Value: 0.000000
Register: 17; Value: 0.000000
Register: 55; Value: 50.008247

I will also provide the results of wireshark, since I can already see an issue there

bkormann, 2018-10-02:
If you refer to the screenshot above, the message with transaction ID 5 is sent but response is returned after 2.5 seconds, but default timeout is set to 0.5 s.

If you have a closer look on the tcp seq/ack. The request of T-ID 5 has SEQ 49 and all queries have a length of 12 bytes. The response of query with T-ID 5 is acknowledges 121, which is query 5, 6, 7, 8, 9, 10 together even though there is no modbus response.

**Details:**
Frame 107: 66 bytes on wire (528 bits), 66 bytes captured (528 bits)
Ethernet II, Src: Beckhoff_32:37:1f (00:01:05:32:37:1f), Dst: JabilCir_0a:19:19 (20:bb:c6:0a:19:19)
Internet Protocol Version 4, Src: 10.203.159.79, Dst: 10.203.159.85
Transmission Control Protocol, Src Port: 53472, Dst Port: 502, Seq: 49, Ack: 53, Len: 12
Modbus/TCP
    Transaction Identifier: 5
    Protocol Identifier: 0
    Length: 6
    Unit Identifier: 255
Modbus
    .000 0011 = Function Code: Read Holding Registers (3)
    Reference Number: 9
    Word Count: 2

Frame 160: 66 bytes on wire (528 bits), 66 bytes captured (528 bits)
Ethernet II, Src: Beckhoff_32:37:1f (00:01:05:32:37:1f), Dst: JabilCir_0a:19:19 (20:bb:c6:0a:19:19)
Internet Protocol Version 4, Src: 10.203.159.79, Dst: 10.203.159.85
Transmission Control Protocol, Src Port: 53472, Dst Port: 502, Seq: 61, Ack: 53, Len: 12
Modbus/TCP
    Transaction Identifier: 6
    Protocol Identifier: 0
    Length: 6
    Unit Identifier: 255
Modbus
    .000 0011 = Function Code: Read Holding Registers (3)
    Reference Number: 11
    Word Count: 2

Frame 198: 66 bytes on wire (528 bits), 66 bytes captured (528 bits)
Ethernet II, Src: Beckhoff_32:37:1f (00:01:05:32:37:1f), Dst: JabilCir_0a:19:19 (20:bb:c6:0a:19:19)
Internet Protocol Version 4, Src: 10.203.159.79, Dst: 10.203.159.85
Transmission Control Protocol, Src Port: 53472, Dst Port: 502, Seq: 73, Ack: 53, Len: 12
Modbus/TCP
    Transaction Identifier: 7
    Protocol Identifier: 0
    Length: 6
    Unit Identifier: 255
Modbus
    .000 0011 = Function Code: Read Holding Registers (3)
    Reference Number: 13
    Word Count: 2

Frame 233: 66 bytes on wire (528 bits), 66 bytes captured (528 bits)
Ethernet II, Src: Beckhoff_32:37:1f (00:01:05:32:37:1f), Dst: JabilCir_0a:19:19 (20:bb:c6:0a:19:19)
Internet Protocol Version 4, Src: 10.203.159.79, Dst: 10.203.159.85
Transmission Control Protocol, Src Port: 53472, Dst Port: 502, Seq: 85, Ack: 53, Len: 12
Modbus/TCP
    Transaction Identifier: 8
    Protocol Identifier: 0
    Length: 6
    Unit Identifier: 255
Modbus
    .000 0011 = Function Code: Read Holding Registers (3)
    Reference Number: 15
    Word Count: 2

Frame 272: 66 bytes on wire (528 bits), 66 bytes captured (528 bits)
Ethernet II, Src: Beckhoff_32:37:1f (00:01:05:32:37:1f), Dst: JabilCir_0a:19:19 (20:bb:c6:0a:19:19)
Internet Protocol Version 4, Src: 10.203.159.79, Dst: 10.203.159.85
Transmission Control Protocol, Src Port: 53472, Dst Port: 502, Seq: 97, Ack: 53, Len: 12
Modbus/TCP
    Transaction Identifier: 9
    Protocol Identifier: 0
    Length: 6
    Unit Identifier: 255
Modbus
    .000 0011 = Function Code: Read Holding Registers (3)
    Reference Number: 17
    Word Count: 2

Frame 307: 66 bytes on wire (528 bits), 66 bytes captured (528 bits)
Ethernet II, Src: Beckhoff_32:37:1f (00:01:05:32:37:1f), Dst: JabilCir_0a:19:19 (20:bb:c6:0a:19:19)
Internet Protocol Version 4, Src: 10.203.159.79, Dst: 10.203.159.85
Transmission Control Protocol, Src Port: 53472, Dst Port: 502, Seq: 109, Ack: 53, Len: 12
Modbus/TCP
    Transaction Identifier: 10
    Protocol Identifier: 0
    Length: 6
    Unit Identifier: 255
Modbus
    .000 0011 = Function Code: Read Holding Registers (3)
    Reference Number: 55
    Word Count: 2

Frame 313: 67 bytes on wire (536 bits), 67 bytes captured (536 bits)
Ethernet II, Src: JabilCir_0a:19:19 (20:bb:c6:0a:19:19), Dst: Beckhoff_32:37:1f (00:01:05:32:37:1f)
Internet Protocol Version 4, Src: 10.203.159.85, Dst: 10.203.159.79
Transmission Control Protocol, Src Port: 502, Dst Port: 53472, Seq: 53, Ack: 121, Len: 13
Modbus/TCP
    Transaction Identifier: 5
    Protocol Identifier: 0
    Length: 7
    Unit Identifier: 255
Modbus
    .000 0011 = Function Code: Read Holding Registers (3)
    [Request Frame: 307]
    Byte Count: 4
    Register 55 (UINT16): 17352
    Register 56 (UINT16): 61848

pboettch, 2018-10-02:
Could you again put the wireshark-dump between backticks to improve readability? Also, it might be better to paste wireshark screenshots or the dump-file directly here. 

For the timeout problem you could try increasing timeouts via

    int modbus_set_response_timeout(mb, 5, 0); // 5 seconds

bkormann, 2018-10-02:
> OK, and if you compare this doing the same with the other modbus-client (on the same host, if possible), is it different?

Can you recommend a modbus tcp client for linux? I don't have any so far.

pboettch, 2018-10-02:
I don't know any modbus client which are not using libmodbus in Linux. But why not trying Ananas in Windows, with wireshark and compare.

From your wireshark log it seems that burst accesses are not supported by the server, too many requests are creating hick-ups, it seems. 

Does the same happen if you read the same register in manner?

You might consider reading all registers at once and just pick up the value you needs:

```C
void func()
{
	modbus_t *mb;
	uint16_t tab_reg[32];
	int rc;
	mb = modbus_new_tcp("10.203.159.85", 502);
	if (modbus_connect(mb) == -1) {
		fprintf(stderr, "Connection failed:\n");
		modbus_free(mb);
		return -1;
	}

	while (1) {
		uint16_t tab_reg[56];
		rc = modbus_read_registers(mb, 1, 56, tab_reg);
		if (c == 56) {

			int tags[10] = {1, 3, 5, 7, 9, 11, 13, 15, 17, 55};

			for (int i = 0; i < 10; i++) {
				float result = 0.0;
				unsigned *value = (unsigned *) &result;

				*value = (tab_reg[tags[i]] << 16) + tab_reg[tags[i] + 1];

				printf("Register: %d; Value: %f\n", tags[i], result);
			}
		} else {
			fprintf(stderr, "Connection failed: %s\n", modbus_strerror(errno));
		}
		sleep(1);
	}
}
```
This code has not been compiled and should _NOT_ be used in production code, it's an example.

karlp, 2018-10-02:
Your sleep is/was at the wrong point.  You were trying to issue back to back requests, without any delay between the 10 reads.  You can see form your trace that your pac3200 stopped responding to them all, and then you sent another blast of 10.  REad the pac3200 docs, even on tcp it probably has al imit of outstanding requests.  Sleep of 100ms between each read will be _far_ more reliable than blast 10, then sleep for 1 second and hope it all came back properly.

stephane, 2019-07-29:
Thank you @pboettch and @karlp for your help.

closed 2019-07-29 by stephane