stephane/libmodbus #453: Usage of "modbus-private.h" in my application program

sritam2, 2018-10-08:
Can we use "modbus-private.h" in our application programs. I am analyzing the structure and content of the data packet that has been received. So, while parsing the data packet, I am using data types defined in "modbus-private.h". Is it allowed and OK ?

Thanks and Regards,
Sritam Paltasingh.

karlp, 2018-10-08:
It's a bad idea, you shouldn't need it, and if you have specific things you think should be exposed, request them explicitly as feature requests.

sritam2, 2018-10-08:
Thank you for your valuable feedback. 
The wrong approach that I was using to extract the header length is:
1. modbus_receive(ctx, query);
//I need to include "modbus-private.h" for this data structure information
2. header_length = ctx->backend->header_length; 
3. address = (query[header_length+1]<<8)+(query[header_length+2]);
4. size = (query[header_length+3]<<8)+(query[header_length+4]);

As per your suggestion now I am planning to replace it by:
1. modbus_receive(ctx, query);
//No need to include "modbus-private.h" now
2. header_length = modbus_get_header_length(ctx);
3. address = (query[header_length+1]<<8)+(query[header_length+2]);
4. size = (query[header_length+3]<<8)+(query[header_length+4]);

"address" is the address sent by the client from where it want to read or write.
"size" is the number of registers/bits it(client) wants to read from the start address or address.

The above code are inside the modbus_server that I am programming.

Do you think this approach is good enough. Is the mechanism used by me to parse "address" and "size" values from the received modbus data packet correct? 
Should I use the array indexes "header_length+1", "header_length+2", "header_length+3", "header_length+4" to extract "address" and "size" values respectively or is there any good alternative to that also?

Please give your valuable suggestion.

Thanks and Regards,
Sritam Paltasingh.

closed 2018-10-29 by sritam2