stephane/libmodbus #454: libmodbus ./configure

baskar1975, 2018-10-17:
Lib version:libmodbus-3.0.6
Host PC :ubuntu 16.1
Target compilation: arm-linux-gnueabihf-gcc
Steps Lib compilation:
1. sudo CC=${PATH} ./configure --host=arm-linux-gnueabihf --enable-static 
{
export PATH="/home/atop_sdk/ti-processor-sdk-linux-am335x-evm-05.00.00.15/linux-devkit/sysroots/x86_64-arago-linux/usr/bin:$PATH"
export  ARCH=arm
export CROSS_COMPILE=arm-linux-gnueabihf-
}
2. make ARCH=arm  CROSS_COMPILE=arm-linux-gnueabihf-  install.
3. atop-client.c compilation
We are getting the following errors,so please correct above steps to solve the issue

"/usr/local/lib/libmodbus.so: file not recognized: File format not recognized"

pboettch, 2018-10-19:
what is atop-client? How is it compiled?

The first line indicate that you are cross-compiling for ARM. Then you do `make install` on your compilation host. That cannot work as it installs the arm-binary-library in /usr/local. Your program is probably a X86-binary.

baskar1975, 2018-10-19:
atop-clien is my modbus rtu client application program.
Target application should work in am335x processor,My host PC is  64 bit AMD ubuntu 16.4 and cross compile (arm-linux-gnueabihf) to work in am335x.
So first i want to compile and install the libmodebus version 3.1.4 in host pc with cross compilation.

Please first tell us the steps of libmodbus cross compilation in host pc which mention above

Regards
R.Baskar

pboettch, 2018-10-23:
Just don't mix up build for different platforms in your environment. Building libmodbus for your host is documented in the README. Please check there first.

closed 2018-11-14 by stephane