stephane/libmodbus #455: modbus_mapping_new() allocating wrong memory?

ci7i2en4, 2018-10-22:
I may have found an issue in the modbus_mapping_new() function (libmodbus v3.1.4). For some reason it doesn't interpret its parameters as described in the documentation.

I have written a Modbus Slave based on the unit-test-server.c example, however, `modbus_mapping_new(0, 0, 7, 0)` does NOT allocate memory for 7 holding registers. As a result I get a SegFault when writing values to `mb_mapping->tab_registers[i] = 28;`.

What really happens can be seen [here](https://stackoverflow.com/questions/52918085/libmodbus-modbus-mapping-new-not-working-correctly?noredirect=1#comment92746769_52918085). The function interprets the parameter "7" as start address for the input bits that I don't even need.

pboettch, 2018-10-22:
It could be that you're mixing up include and library files from different versions. Like in `/usr/include/` you have a modbus.h from 3.0.x which is used during compilation and when you run your binary it uses a library you compiled locally.

Actually I think it is the inverse: in /usr/lib you have an old version of libmodbus (dynamic, a .so-file) and you compile your application with modbus.h from the source-tree.

To check, run `ldd <binary` and see to which libmodbus.so it refers.

ci7i2en4, 2018-10-23:
You were right: Somehow two different versions of libmodbus seem to have existed on my system. According to apt I had installed 3.0.6 from the official Ubuntu packet sources. However, the LIBMODBUS_VERSION_STRING variable contained "3.1.4". 

So I purged the old version 3.0.6 from the system and reinstalled 3.1.4 after cloning it from the repository. Now everything works as expected again. Thank you for your help!

closed 2018-10-23 by ci7i2en4