stephane/libmodbus #456: Connection time out - rs485

MR-microland, 2018-10-24:
hi i am using this code to read data from modbus simulator form pc and running C code on linux machine.
`#include <stdio.h>
#include <stdlib.h>
#include <errno.h>
#include <stdbool.h>

#include <modbus.h>

int main()
{
  modbus_t *ctx = 0;

  //
  // create a libmodbus context for RTU
  // doesn't check if the serial is really there
  //
  ctx = modbus_new_rtu("/dev/ttyS0", 115200, 'N', 8, 1);

  if (ctx == 0) {

    fprintf(stderr, "Unable to create the libmodbus context\n");
    return -1;

  } else {
    struct timeval old_response_timeout;
    struct timeval response_timeout;

    // enable debug
    modbus_set_debug(ctx, true);

    // initialize timeouts with default
    modbus_get_response_timeout(ctx, &old_response_timeout);
    response_timeout = old_response_timeout;

    // set the message and charcater timeout to 2 seconds
    response_timeout.tv_sec = 2;
    modbus_set_response_timeout(ctx, &response_timeout);
    modbus_set_byte_timeout(ctx, &response_timeout);

  }

  // try to connet to the first DZT on the line
  // assume that line address is 1, the default
  // send nothing on the line, just set the address in the context
  if(modbus_set_slave(ctx, 1) == -1) {
    fprintf(stderr, "Didn't connect to slave/n");
    return -1;
  }

  // establish a Modbus connection
  // in a RS-485 context that means the serial interface is opened
  // but nothing is yet sent on the line
  if(modbus_connect(ctx) == -1) {

    fprintf(stderr, "Connection failed: %s\n", modbus_strerror(errno));
    modbus_free(ctx);
    return -1;

  } else {

    int nreg = 0;
    uint16_t tab_reg[32];

    fprintf(stderr, "Connected\n");

    //
    // read all registers in DVT 6001
    // the function uses the Modbus function code 0x03 (read holding registers).
    //
    nreg = modbus_read_registers(ctx,0,1,tab_reg);

    if (nreg == -1) {

       fprintf(stderr, "Error reading registers: %s\n", modbus_strerror(errno));
       modbus_close(ctx);
       modbus_free(ctx);

       return -1;

    } else {
       int i;

       // dump all registers content

       fprintf (stderr, "Register dump:\n");
       for(i=0; i < nreg; i++)
         printf("reg #%d: %d\n", i, tab_reg[i]);

       modbus_close(ctx);
       modbus_free(ctx);

       return 0;
    }
  }
}`

i am getting connection time out for this 
`Opening /dev/ttyS0 at 115200 bauds (N, 8, 1)
Connected
[01][03][00][00][00][01][84][0A]
Waiting for a confirmation...
ERROR Connection timed out: select
Error reading registers: Connection timed out
`

karlp, 2018-10-24:
connection time out means just that.  Check:
- [ ] your wiring (are A/B right? try just swapping them, some companies use different markings)
- [ ] your serial port (is it the right one? is it automatically handling rs485 for you?)
- [ ] baud rate (modbus recommended default is 19200, check!)
- [ ] parity (modbus recommend default is even, check!)
- [ ] unit address (don't rely on defaults)
- [ ] does it respond to function code 3? (while it's desired for themto reply with ILLEGAL FUNCTION, they can also just ignore you)
- [ ] does it respond at address 0? (while it's desired for them to reply with BAD ADDRESS, they can also just ignore you)

finally, you're reading only 1 register, then attempting to print 32, you probably don't want to do that, but it's not causing your timeouts.

closed 2018-10-24 by MR-microland

MR-microland, 2018-10-25:
=>I have DB9F to TB converter from moxa from that i am connecting to rs 485 port of 2 wire D+ and D-
=>Function code i have set to 3 in the modbus slave simulator.
=>Buad rate is flexible for this port.

applea9, 2021-03-25:
@MR-microland I have the same problem now, how did you slove it? thanks for your reply.