stephane/libmodbus #458: Security of modbus data packets against "man-in-the-middle" attacks

sritam2, 2018-10-29:
After successful execution of "modbus_new_tcp" a new context gets created. But the context is on an open port. So, it is vulnerable to attacks. Is there any mechanism to make the modbus data transfers between client and server secure so that it is not vulnerable to attacks such as "man-in-the-middle" attacks.

As far as I know, libmodbus does not offer any security mechanism for modbus data packets.
Please help with your valuable advise.

Thanks and Regards,
Sritam Paltasingh.

watsocd, 2018-10-29:
The MODBUS protocol does not implement any kind of security.

 

The best “security” I have seen is to program your slave clients to only respond to specified MODBUS masters by IP address. If a poll request comes from another IP address, program the slave to not respond.

 

Obviously MODBUS is not the right protocol to use on the public Internet if security is a big concern. Man-in-the-middle attacks are extremely rare on corporate/private LANs/WANs.

 

Chuck Watson BS, CEM, CMVP

Direct USA 724-799-1529

mhei, 2018-10-29:
Wouldn't it make more sense to reject the incoming connection from an undesired IP address instead of not answering the requests?
Apart from this, you could use something like [stunnel](https://www.stunnel.org/) to strip off a TLS encryption before passing the connection down to your Modbus slave.

jcwren, 2018-10-29:
If you're actually that concerned about security, all your nodes should be
on a VPN. And if they're at different physical sites, they'd ideally be on
a VPN unique to that site, so that if one gets compromised, the others are
still safe.

--jc

Save a life; adopt a shelter animal.

sritam2, 2018-10-29:
There is also a specification by modbus community known as mbap/TLS/TCP which specifies the security of modbus data packets.
The modbus specification without security is called mbap(modbus application protocol) and the modbus specification with security is called as mbaps(modbus security application protocol). 

What I understand is that Libmodbus is implementation of mbap and not mbaps. Correct me if I am wrong.

Please help me out with information if you know any open source implementation of mbaps.

Thanks and Regards,
Sritam Paltasingh.

watsocd, 2018-10-29:
This only works if you can find slave devices that will support the secure protocol. The advantage of MODBUS is a plethora of slave devices on the market using the old protocols. 

 

If you control the programming of both the slaves and the masters and are worried about security, why are you using MODBUS at all?

sritam2, 2018-10-30:
Actually the client devices(masters) are done by our customers and they will ping our server(slave) via modbus. the clients shall be at different physical sites and would connect the server via internet. that is why i am concerned about security.

stephane, 2018-11-14:
This discussion should take place on the mailing list not on the issue tracker.

closed 2018-11-14 by stephane