stephane/libmodbus #461: modbus_write_register(...) : inconsistent type of last input-parameter

dreamflow, 2018-11-19:
in [modbus.c](https://github.com/stephane/libmodbus/blob/master/src/modbus.c) in the function [modbus_write_register(...)](https://github.com/stephane/libmodbus/blob/master/src/modbus.c#L1280) the type **`int`** of the last
input-parameter **`value`** is inconsistent compared to the last parameters of the
following functions from [modbus.c](https://github.com/stephane/libmodbus/blob/master/src/modbus.c)
```c
int modbus_write_register      (modbus_t *ctx, int addr        ,        int     value); /* <-- */
int modbus_write_registers     (modbus_t *ctx, int addr, int nb, const uint16_t *src );
int modbus_read_registers      (modbus_t *ctx, int addr, int nb,       uint16_t *dest);
int modbus_read_input_registers(modbus_t *ctx, int addr, int nb,       uint16_t *dest);
```

for consistency **`int value`** should be **`const uint16_t value`** ?

```c
int modbus_write_register      (modbus_t *ctx, int addr        , const uint16_t value);
```

pboettch, 2018-11-19:
The reason why there is no const for this primitive, non-reference parameter could this one:

[Is it better to remove “const” in front of “primitive” types used as function parameters in the header?](https://stackoverflow.com/a/46292715/880584)

pboettch, 2018-11-19:
Yeah, sorry, I read your issue as if you were thinking about adding const. Indeed it seems `uint16_t` would be more appropriate.

closed 2019-07-29 by stephane

stephane, 2019-07-29:
Merged. Thank you.

commit 862af10b (2019-07-29): Change arg value from int to const uint16_t in modbus_write_register

Closes #461