stephane/libmodbus #466: Error reading DWord type (32 bit) device

wayneenterprise1, 2018-12-28:
Hi! I'm trying to read data from high-speed counter HC202 (DWord, see the screenshot below) inside Delta AS228T PLC using libmodbus TCP (v3.0.6):

```
uint16_t test_data[10];
modbus_read_registers(mb, 64715, 1, test_data);
```

But it returns "Invalid data" error. Reading Word or Bit type devices works fine, but it fails with DWord type.

[image]

I think it's because libmodbus cant't take more data than requested. In my case I read one register but I would actually get 2 registers returned (one 32 bit value) instead of the usual one 16 bit register.

I just forced libmodbus to work fine with the following code:

```
uint16_t test_data[2];
num = modbus_read_registers(mb, 64714, 1, test_data);
Form1->Caption = (long)test_data[0] << 16 | test_data[1];
```

However, I had to remove the following lines from modbus.c and reassemble the library:

```
1135. rc = check_confirmation(ctx, req, rsp, rc);
1136.        if (rc == -1)
1137.            return -1;
```

pboettch, 2018-12-28:
Could you re-run your test-case with `modbus_set_debug(ctx, TRUE);` and paste the output here (without the check_confirmation commented)?

For your device, how would you read address higher than 65535?

wayneenterprise1, 2018-12-28:
Here you are:

`[00][01][00][00][00][06][FF][03][FC][CA][00][01]
Waiting for a confirmation...
<00><01><00><00><00><07><FF><03><04><00><04><8F><5A>
Message length not corresponding to the computed length (13 != 11)
0x1 0 (-1)`

Well, check this link to ensure that addresses are not actually higher then 65535: https://www.kepware.com/en-us/support/knowledge-base/2008/modbus-5-digit-addressing-versus-6-digit-addressin/

pboettch, 2018-12-28:
OK, that basically confirms what I said in my mail. A function modbus_read_registers() with a `uint32_t[]`-data-array would be the right way to do it, but, is this modbus-standard-conform or is this a custom extension made by this manufacturer?

pboettch, 2018-12-28:
I just checked that this technically feasible to add 

```C
MODBUS_API int modbus_read32_registers(modbus_t *ctx, uint16_t addr, int nb, uint32_t *dest);
MODBUS_API int modbus_read32_input_registers(modbus_t *ctx, uint16_t addr, int nb, uint32_t *dest);

MODBUS_API int modbus_write32_registers(modbus_t *ctx, uint16_t addr, int nb, const uint32_t *data);
MODBUS_API int modbus_write32_register(modbus_t *ctx, uint16_t reg_addr, int32_t value);
```
while keeping the compatibility efforts in the library to a minimum, however there would be some changes to core functions which would require testing.

pboettch, 2018-12-28:
In the modbus frame the number of values are encoded, _not_ the number of bytes.

```C
static int _modbus_rtu_build_request_basis(..)
{
    [..]
    req[4] = nb >> 8;
    req[5] = nb & 0x00ff;
    [..]
}
```
The non-compliance-problem of the device is that it return 4 bytes per register and not 2. Doing what you suggest `modbus_read_registers(mb, addr, 2, res);` would return 8 bytes in @wayneenterprise1's case.

wayneenterprise1, 2018-12-28:
@pboettch, thanks a lot! It is so-called Enron/Daniels mode which is non-standard Modbus. What should I do next?

pboettch, 2018-12-28:
I'm not the maintainer of this library, so I can only add my opinion: I'd add support for the Enron/Daniels mode. Whether it is an official standard or not, if it is widely supported (and documented) it would be justified for me.

As to adding support for this here, there multiple possibilities: from you check out the code and see what you can do to add it yourself and when done (with the community's support), issue a pull-request to get (more) feedback. Or you can ask someone else to do it if you are in a hurry (or missing the required skills). I could do it, but I'm quite busy at the moment.

The core functions I mentioned above are `read_registers()`, `check_confirmation()`, `write_register()` and `modbus_write_registers`. They need to learn about register-sizes other than 2-bytes. Then code the functions I mentioned above and documents them and put in place a unit-test-case.

mhei, 2018-12-28:
@wayneenterprise1: After reading the article I have the impression that this type of access was required for older products and that newer ones still support it for backwards compatibility? So can you just disable this mode?

karlp, 2018-12-28:
Ok, so that device is just completely non-compliant.  I've got no problems adding support for this non-modbus/modbus, but _definitely_ it needs a different name than "read32" which would be terribly confusing for all the _other_ compliant implementations that use 2 registers for 32bit values :)

(and enron/daniels should be taken out and shot for this botchjob, everyone else just did it, instead of thinking they needed magic modes)

(Finally, does the device in question actually have a mode to choose that actually enables compliant mode, instead of enron/daniels mode?)

wayneenterprise1, 2018-12-28:
@pboettch, you're right, I'm both in a hurry and not experienced enough but I'll try to deal with it.
@mhei, @karlp, no, I'm unable to do it because of the architecture of the PLC (Programmable logic controller) - there is no such an opportunity provided.
@karlp, I will be very grateful to you for any support.

s-m-r, 2018-12-28:
I have written a lot of embedded Modbus code and from what I read in the Enron/Daniels document, I am glad I have not had the displeasure of interfacing with one.

wayneenterprise1, 2018-12-29:
Adding 2 to computed response length and 1 to number of recieved values for `check_confirmation` function is enough to make it work fine.

wayneenterprise1, 2018-12-29:
Adding 2 to computed response length and subtracting 1 from number of recieved values for `check_confirmation` function is enough to make it work fine.

stephane, 2019-07-28:
I'm reluctant to add support for such non compliant devices excepted if very common.
Do you seem it's the case?

wayneenterprise1, 2019-07-29:
I'm not sure if it is widely used, but modpoll supports it, and the model of PLC that I had to work with was first released in 2017 (quite modern). So this kind of registers seems to be relevant.

Flow86, 2022-11-03:
Hi, I also encountered such a device.

Since I only read from such a device, I only had to change check_confirmation a bit (i.e by ignoring the remainder data.)

My modbus client still requests the compliant register amount - for my use case this works well, but probably can fail (i.e if you request the "last" two registers to read a 32bit value)

```
diff --git a/src/modbus.c b/src/modbus.c
index e13bd0d..3c6dc5f 100644
--- a/src/modbus.c
+++ b/src/modbus.c
@@ -551,7 +551,7 @@ static int check_confirmation(modbus_t *ctx, uint8_t *req,
     }
 
     /* Check length */
-    if ((rsp_length == rsp_length_computed ||
+    if ((rsp_length >= rsp_length_computed ||
          rsp_length_computed == MSG_LENGTH_UNDEFINED) &&
         function < 0x80) {
         int req_nb_value;
@@ -607,6 +607,14 @@ static int check_confirmation(modbus_t *ctx, uint8_t *req,
 
         if (req_nb_value == rsp_nb_value) {
             rc = rsp_nb_value;
+        } else if (req_nb_value < rsp_nb_value) {
+            if (ctx->debug) {
+                fprintf(stderr,
+                        "Non compliant response: Got more data than we have requested. (%d != %d)\n",
+                        rsp_nb_value, req_nb_value);
+            }
+            /* lets use the expected value count as a result, and ignore the remainder */
+            rc = req_nb_value;
         } else {
             if (ctx->debug) {
                 fprintf(stderr,
```

karlp, 2022-11-03:
@Flow86 how does ignoring the remainder work?  the problem in enron/daniels is that you receive request_reg_count * 4 bytes, not request_reg_count * 2 bytes.  And you _need_ that data, because you're getting 32bit results, not 16bit results.  You can't just ignore the remainder.

Flow86, 2022-11-03:
@karlp well, for my use case this works: I read only one value (i.e 2 16bit registers) from the device with a call, so calling `modbus_read_registers` with `2` - from that non compliant device I get 4 16bit registers back but I don't care about the 3rd&4th register of that response.

But to properly implement an elron/daniels mode would be difficult I think, since you can't know which registers are bigger from the start of the request.