stephane/libmodbus #469: Consequent handling of EINTR

franzhollerer, 2019-02-20:
IMHO handling of EINTR needs more attention. Nearly every system call might return EINTR, and should be restarted if applicable. Skimming through the code I miss this for several system calls:

- send()
- recv()
- connect()
- select()
- ...

mhei, 2019-02-20:
I agree, that nearly every system call might return EINTR and that we should handle this in a consequent manner. But I'd not agree to restart the system call in library. Handling of EINTR should be left to the application. For example, your application will receive a EINTR in case you hit Ctrl-C while in recv. Your (the users) goal is to terminate the application but when you restart the recv in library code, your application will simply not exit.
So IMHO the library should simply return to caller, leave errno as is and thus give the application control howto proceed.

franzhollerer, 2019-02-20:
If you keep the standard SIGINT handler, restarting the function is fine as the default handler terminates the program anyway.

If you install your own handler you must have a clear strategy on how to shut down your application cleanly. I think relying on EINTR is not sufficient in this case, and you will get more problems than you solve.

The other problem is that we cannot handle a correct restart of the involved system calls at application level. E.g if you look at function _connect().

```c
static int _connect(int sockfd, const struct sockaddr *addr, socklen_t addrlen,
                    const struct timeval *ro_tv)
{
    int rc = connect(sockfd, addr, addrlen);

#ifdef OS_WIN32
    int wsaError = 0;
    if (rc == -1) {
        wsaError = WSAGetLastError();
    }

    if (wsaError == WSAEWOULDBLOCK || wsaError == WSAEINPROGRESS) {
#else
    if (rc == -1 && errno == EINPROGRESS) {
#endif
        fd_set wset;
        int optval;
        socklen_t optlen = sizeof(optval);
        struct timeval tv = *ro_tv;

        /* Wait to be available in writing */
        FD_ZERO(&wset);
        FD_SET(sockfd, &wset);
        rc = select(sockfd + 1, NULL, &wset, NULL, &tv);
        if (rc <= 0) {
            /* Timeout or fail */
            return -1;
        }

        /* The connection is established if SO_ERROR and optval are set to 0 */
        rc = getsockopt(sockfd, SOL_SOCKET, SO_ERROR, (void *)&optval, &optlen);
        if (rc == 0 && optval == 0) {
            return 0;
        } else {
            errno = ECONNREFUSED;
            return -1;
        }
    }
    return rc;
}
```

This function calls connect() followed by select(). Both may fail with EINTR. If the appliction gets EINTR from a call to modbus_connect(), it has no idea if connect() or socket() was interrupted, nor on how to restart the interrupted system call. Therefore, I think EINTR must be handled within the library, at the place where it occures. The application cannot handle it.

As far as I can see none of the libmobus function blocks for ever. Thus restarting the system call should be not a great deal. You still have the timeouts to keep the application reacting on other events, e.g. a shutdown flag set by a custom SIGINT handler.

IMHO EINTR is archaic. Most systems nowadays support SA_RESTART, but for portability the library should deal with it. The application has no idea which of the involved system was interrupted, nor how to recover from this error.