stephane/libmodbus #472: ".in'ig.status:" while running ./configure on Ubuntu18.10

ccf19881030, 2019-03-15:
Whe I run Host PC :ubuntu 18.10

1、clone the repo

2、checkout branch v3.0.X 

3、run autogen.sh
./autogen.sh
get this error:
`libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
libtoolize:   error: linking '/usr/share/libtool/build-aux/ltmain.sh' to 'build-aux/' failed
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize:   error: linking '/usr/share/aclocal/libtool.m4' to 'm4/' failed
libtoolize:   error: linking '/usr/share/aclocal/ltoptions.m4' to 'm4/' failed
libtoolize:   error: linking '/usr/share/aclocal/ltsugar.m4' to 'm4/' failed
libtoolize:   error: linking '/usr/share/aclocal/ltversion.m4' to 'm4/' failed
libtoolize:   error: linking '/usr/share/aclocal/lt~obsolete.m4' to 'm4/' failed
autoreconf: libtoolize failed with exit status: 1

--------------------------
Running autoreconf failed.
--------------------------`
4、build from source fails on Ubuntu18.10
./configure --prefix=/usr/local
get this error:
`VirtualBox:/mnt/Windows_Share/GithubProjects/libmodbus/libmodbus$ ./configure --prefix=/usr/local/
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
/bin/bash: /mnt/Windows_Share/GithubProjects/libmodbus/libmodbus/build-aux/missing: No such file or directory
configure: WARNING: 'missing' script is too old or missing
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... none
checking for gcc option to accept ISO C99... none needed
checking for gcc option to accept ISO Standard C... (cached) none needed
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files... no
checking whether make supports nested variables... (cached) yes
[... 111 lines cut ...]
checking if the linker (/usr/bin/ld -m elf_x86_64) is GNU ld... yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking for g++ option to produce PIC... -fPIC -DPIC
checking if g++ PIC flag -fPIC -DPIC works... yes
checking if g++ static flag -static works... yes
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking for int64_t... yes
checking for size_t... yes
checking for ssize_t... yes
checking for uint16_t... yes
checking for uint32_t... yes
checking for uint8_t... yes
checking winsock2.h usability... no
checking winsock2.h presence... no
checking for winsock2.h... no
checking whether TIOCSRS485 is declared... yes
checking whether TIOCM_RTS is declared... yes
checking that generated files are newer than configure... done
configure: creating ./config.status
.in'ig.status: error: cannot find input file: ``
I can't build the libmodbus sources on my Ubuntu18.10 in Oracle VM VirtualBox 6.0.2 r128162,I can't find the reason why ,thank you!

ccf19881030, 2019-03-15:
1、".in'ig.status:" while running ./configure on Ubuntu18.10
2、autoreconf: libtoolize failed with exit status: 1 when run ./autogen.sh

mellaigh, 2019-05-23:
Hi, the same problem encountered @ Debian9.

mellaigh, 2019-05-23:
Hi fixed. 
git clone in Linux environment, don't do it in windows. 
CR-LF makes this problem.

ccf19881030, 2019-06-04:
@mellaigh  Thanks,Now I have git clone in Linux environment,this Solved my problem.

closed 2019-06-04 by ccf19881030

qqq89513, 2021-05-03:
@mellaigh Thanks for the info.
Just to remind that if you `git clone` in a host shared folder in a Linux environment, it results the same error, 
where the host is Windows.