stephane/libmodbus #482: how to add libmodbus to cmake project

magiino, 2019-05-15:
I would like to include this lib to my CMAKE [project](https://github.com/magiino/I40Barman). I have this library as git submodule in lib/libmodbus (not yet pushed on git) or with ExternalProject_Add(). What should I add to *CMakeLists.txt*? I do not know find any useful advices anywhere, how should be this done.

i have already tried ExternalProject_Add() but it does not worked.
cmake code below --->
```
cmake_minimum_required(VERSION 3.13)
project(I40Barman)

set(CMAKE_CXX_STANDARD 11)
set(INCLUDE_FILES ${PROJECT_SOURCE_DIR}/include)
set(SOURCE_FILES ${PROJECT_SOURCE_DIR}/src)

find_package(open62541 0.4.0 REQUIRED COMPONENTS FullNamespace Methods Subscriptions Encryption Discovery DiscoveryMulticast)
find_package(nlohmann_json 3.2.0 REQUIRED)

include(ExternalProject)
set(EXTERNAL_INSTALL_LOCATION ${CMAKE_BINARY_DIR}/external)

ExternalProject_Add(
    project_libmodbus
    GIT_REPOSITORY https://github.com/stephane/libmodbus.git
    GIT_TAG master
    CMAKE_ARGS -DCMAKE_INSTALL_PREFIX=${EXTERNAL_INSTALL_LOCATION}
)

include_directories(${EXTERNAL_INSTALL_LOCATION}/include)
link_directories(${EXTERNAL_INSTALL_LOCATION}/lib)

include_directories(
        ${INCLUDE_FILES}
        ${SOURCE_FILES}
)

file(GLOB_RECURSE all_SRCS
     "${INCLUDE_FILES}/*.h"
     "${INCLUDE_FILES}/*.hpp"
     "${SOURCE_FILES}/*.cpp"
     "${SOURCE_FILES}/*.c"
)

add_executable(
        ${PROJECT_NAME}
        ${all_SRCS}
)

add_dependencies(${PROJECT_NAME} project_libmodbus)

target_link_libraries(${PROJECT_NAME} PRIVATE nlohmann_json::nlohmann_json libmodbus open62541::open62541)
```

stephane, 2019-07-28:
Questions should be posted on libmodbus mailing list.

closed 2019-07-28 by stephane

magiino, 2020-03-26:
Thanks to `rightback`, who provided me a solution !!!

**The solution covers configuring build of make library(libmodbus) with CMake**

**Project structure:**
project dir/
| -- **CMakeLists.txt**
| -- **modbus.cmake**
| -- src dir/
| -- | -- ... source files
| -- include dir/
| -- | -- ... header files
| -- lib dir/
| -- | -- json (git submodule)
| -- | -- **libmodbus** (git submodule)

**Project CMakeLists.txt**
```
project(process-cell)

# Json library
set(JSON_BuildTests OFF CACHE INTERNAL "")
add_subdirectory(lib/json)

# Libmodbus library
include(modbus.cmake)

# Find all source files with file extension .c/.cpp inside src folder
file(GLOB_RECURSE PROJECT_SRCS "src/*.c" "src/*.cpp")

# Add source files to project
add_executable(
    ${PROJECT_NAME}
    ${PROJECT_SRCS}
)

# Include project header files
target_include_directories(
    ${PROJECT_NAME}
    PUBLIC include
)

target_link_libraries(
    ${PROJECT_NAME}
    PRIVATE open62541-cpp-api
    PRIVATE nlohmann_json::nlohmann_json
    PRIVATE modbus
)
```

**Libmodbus setup modbus.cmake**
```
#
# Builds the modbus library
#
# Outputs the following target:
#   modbus
#
include(ExternalProject)
set(MODBUS_DIR ${CMAKE_CURRENT_SOURCE_DIR}/lib/libmodbus)
set(MODBUS_BIN ${CMAKE_CURRENT_BINARY_DIR}/libmodbus)
set(MODBUS_STATIC_LIB ${MODBUS_BIN}/lib/libmodbus.a)
set(MODBUS_INCLUDES ${MODBUS_BIN}/include)

file(MAKE_DIRECTORY ${MODBUS_INCLUDES})

ExternalProject_Add(
  libmodbus
  PREFIX ${MODBUS_BIN}
  SOURCE_DIR ${MODBUS_DIR}
  DOWNLOAD_COMMAND cd ${MODBUS_DIR} && git clean -dfX && ${MODBUS_DIR}/autogen.sh
  CONFIGURE_COMMAND ${MODBUS_DIR}/configure --srcdir=${MODBUS_DIR} --prefix=${MODBUS_BIN} --enable-static=yes --disable-shared
  BUILD_COMMAND make
  INSTALL_COMMAND make install
  BUILD_BYPRODUCTS ${MODBUS_STATIC_LIB}
)

add_library(modbus STATIC IMPORTED GLOBAL)

add_dependencies(modbus libmodbus)

set_target_properties(modbus PROPERTIES IMPORTED_LOCATION ${MODBUS_STATIC_LIB})
set_target_properties(modbus PROPERTIES INTERFACE_INCLUDE_DIRECTORIES ${MODBUS_INCLUDES})
```

gocarlos, 2020-03-27:
we will have libmodbus soon as conan package: https://github.com/conan-io/conan-center-index/pull/821

Aalococo, 2020-12-22:
the proposed solution seems to work fine but i do get an error:

lib/libmodbus/src/modbus.h:21:10: fatal error: modbus-version.h: No such file or directory
   21 | #include "modbus-version.h"

How can i tell cmake to fill the included modbus-version.h.in ?