stephane/libmodbus #497: How to generate static lib for linux

snahmad, 2019-10-14:
- Tell me steps.
 - for any linux platform

nwoetzel, 2019-11-11:
My impression is, that the documentation https://github.com/stephane/libmodbus#installation is good.
Have you tried to follow these steps? When using configure on linux, you might need to add "--enable-static --disable-shared" to have the static libs built - but these are standard configure flags, which might not need to be documented explicitly. `./configure --help` lists available options.

kuhrusty, 2019-12-09:
Yeah, this is working for us on Debian 8 or 9 to build a static library *instead* of a shared library:

    $ ./configure --with-pic --enable-static --enable-shared=no

stephane, 2024-10-22:
https://github.com/stephane/libmodbus/wiki/Static-library

closed 2024-10-22 by stephane