stephane/libmodbus #498: SEGV found in unit-test-server.c (v3.1.6)

Rrooach, 2019-10-17:
Hello, I found a __SEGV__  in stephane/libmodbus/unit-test-server.c.

__Below are Steps Followed to Reproduce Crash__
Download latest source code from: stephane/libmodbus/unit-test-server.c, compiled with clang and ASAN```export CFLAGS="-g -fsanitize=address" LDFLAGS="-fsanitize=address"```before make

__Input Data__
[crash.zip](https://github.com/stephane/libmodbus/files/3737263/crash.zip)

__Crash Output__ 
```
The client connection from 0.0.0.0 is accepted
Waiting for an indication...
<00><00><00><00><00><0D><FF><17><01><60><00><20><00><00><00><01><02><12><34>
ASAN:DEADLYSIGNAL
=================================================================
==6401==ERROR: AddressSanitizer: SEGV on unknown address 0x605ffffffe20 (pc 0x7f8c9bc3e84e bp 0x7fff63479810 sp 0x7fff634795a0 T0)                                                                                 ==6401==The signal is caused by a WRITE memory access.                                                                                                                                                                 #0 0x7f8c9bc3e84d in modbus_reply /root/temp/iec/untest/libmodbus/src/modbus.c:981:44
    #1 0x511ddd in main /root/temp/iec/untest/libmodbus/tests/unit-test-server.c:183:14
    #2 0x7f8c9ad3d82f in __libc_start_main /build/glibc-LK5gWL/glibc-2.23/csu/../csu/libc-start.c:291
    #3 0x41a168 in _start (/root/temp/iec/untest/libmodbus/tests/unit-test-server+0x41a168)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /root/temp/iec/untest/libmodbus/src/modbus.c:981:44 in modbus_reply
==6401==ABORTING
```

psychon, 2024-09-22:
I fail to reproduce this with commit cbb0ab9c8fc6e8e3d3.
- Clone git, `cd` to git folder
- `export CFLAGS="-g -fsanitize=address" LDFLAGS="-fsanitize=address"`
- `./autogen.sh && ./configure && make`
- Run `./tests/unit-test-server`
- In another terminal, download the zip file, unzip it, run `netcat localhost 1502 < 1`
I get the following output on the `unit-test-server` and nothing else happens:
```
Client connection accepted from 127.0.0.1.
Waiting for an indication...
<00><00><00><00><00><0D><FF><17><01><60><00><20><00><00><00><01><02><12><34>
Illegal data read address 0x180 or write address 0x0 write_and_read_registers
[00][00][00][00][00][03][FF][97][02]
Waiting for an indication...
```

psychon, 2024-09-22:
Well, okay, this reproduces with v3.1.6, as said by the title. So I can do a `git bisect`. And this ends where basically all bisects end:
```
commit b4ef4c17d618eba0adccc4c7d9e9a1ef809fc9b6 (HEAD)
Author: Michael Heimpold <mhei@heimpold.de>
Date:   Sat Jan 8 20:00:50 2022 +0100

    modbus_reply: fix copy & paste error in sanity check (fixes #614)
    
    While handling MODBUS_FC_WRITE_AND_READ_REGISTERS, both address offsets
    must be checked, i.e. the read and the write address must be within the
    mapping range.
    
    At the moment, only the read address was considered, it looks like a
    simple copy and paste error, so let's fix it.
    
    Signed-off-by: Michael Heimpold <mhei@heimpold.de>

 src/modbus.c | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)
```