stephane/libmodbus #513: [patch] Install the NEWS and AUTHORS files as part of the documentation

richardash1981, 2020-01-05:
## libmodbus version

 3.1.6

## OS and/or distribution

 Linux Gentoo

## Environment

All (documentation)

## Description

 libmodbus 3.1.6 on Linux does not install the NEWS and AUTHORS files as part of it's installed documentation (in /usr/share/doc/libmodbus-3.1.6/). This is a problem because they may be needed, and are referred to from other files which are installed in that directory.

Michael Weber patched this in the Gentoo package for 3.1.4 (https://gitweb.gentoo.org/repo/gentoo.git/tree/dev-libs/libmodbus/files/libmodbus-3.1.4-doc.patch) but the patch has not made it upstream. It's only a one-line change to append `AUTHORS NEWS` to the value of `dist_doc_DATA`

```
--- libmodbus-3.1.4/Makefile.in
+++ libmodbus-3.1.4/Makefile.in
@@ -379,7 +379,7 @@
 pkgconfigdir = $(libdir)/pkgconfig
 pkgconfig_DATA = libmodbus.pc
 EXTRA_DIST = libmodbus.pc.in
-dist_doc_DATA = MIGRATION README.md
+dist_doc_DATA = MIGRATION README.md AUTHORS NEWS
 SUBDIRS = src doc $(am__append_1)
 all: config.h
 	$(MAKE) $(AM_MAKEFLAGS) all-recursive
```
(apparently you cannot attach a patch file to this tracker, so content pasted or download from URL above).

pboettch, 2020-01-06:
As written on the mailing, it is much easier for everyone if you could fork, commit, push and pull-request your changes. Try it, it is easy and cool :sunglasses: .

lssp-maker, 2020-07-15:
大佬就是不一样

PR #545 (2021-04-29): Install NEWS and AUTHORS files with library, ship LICENSE with tests
Ensure that relevant documentation files are shipped and installed with the library and the tests.

Install NEWS and AUTHORS files to doc dir alongside MIGRATION and README.md. Patched by Michael Weber in the Gentoo package for 3.1.4
Fixes #513 

Include the test license in the source tarball.
Fixes #542
commit 26c85163 (2021-04-29): Install the NEWS and AUTHORS files

Fix #513
closed 2021-04-29 by stephane