stephane/libmodbus #514: Escalation of privileges issue

ezioyan, 2020-01-09:
Hello, we found an **Escalation of Privileges** issue in stephane/libmodbus (v3.1.6). 

Send a carefully constructed message with  `FC=0x03`  to server:

```
<00><01><00><00><1D><00><FF><03><00><00><00><0A><00><02><00><00><00><11><FF><10><00><00><00><05><0A><00><00><00><01><00><02><00><03><00><04> 
```

Libmodbus broke it into two messages with `FC=0x03`  and `FC=0x10`, respectively. Server executed these two instructions:

message 1:

```
Waiting for an indication...
<00><01><00><00><1D><00><FF><03><00><00><00><0A>
[00][01][00][00][00][17][FF][03][14][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00][00]
```

message 2:

```
Waiting for an indication...
<00><02><00><00><00><11><FF><10><00><00><00><05><0A><00><00><00><01><00><02><00><03><00><04>
[00][02][00][00][00][06][FF][10][00][00][00][05]
```

In this case, **we wrote registers under read permission**.

In theory, a  `FC 0x03`  message's length field should have a fixed value of 6. That is to say, any message with a length field value greater than 6 should be identified as an invalid one. We noticed that libmodbus does not check length field, so there must be many other issues of this kind.

We suggest **checking length field** immediately a message comes and dropping messages with invalid length field, so as to avoid this kind of issues.

JoelStienlet, 2020-01-09:
Hello,
I've not been able to reproduce this with version 3.1.6. I used the RTU backend, with a virtual serial port under linux: socat -d -d pty,raw,echo=0 pty,raw,echo=0
I wrote the message on the serial port like this:
char message[] = {0x00,0x01,0x00,0x00,0x1D,0x00,0xFF,0x03,0x00,0x00,0x00,0x0A,0x00,0x02,0x00,0x00,0x00,0x11,0xFF,0x10,0x00,0x00,0x00,0x05,0x0A,0x00,0x00,0x00,0x01,0x00,0x02,0x00,0x03,0x00,0x04 };
write (fd, message, sizeof(message)); 

Here what I get with modbus_set_debug(ctx, TRUE);
Waiting for an indication...
<00><01><00><00><1D><00><FF><03>
ERROR CRC received 0xFF03 != CRC calculated 0x348B

For RTU the first byte is the destination address: <00> would be broadcast.
Then we have the function code: <01> and here I don't get why you say it is 0x03?

When you say "we wrote registers under read permission." I suppose you actually meant that you wrote registers while issuing a read command, as there is no notion of "permission" included in the specification of the protocol.

ezioyan, 2020-01-10:
Sorry, I didn't make it clear. We used modbus TCP protocal, so the starting 7 bytes is  header and <03> is the function code. Using a TCP backend can reproduce this issue.
Receiving a Read message but actually executing Read and Write can be a dangerous action.

JoelStienlet, 2020-01-10:
Thank you for this precision!
So the message can be decomposed as follows:

A first header, with transaction id =1 followed by 0x1D=29d bytes: 
<00><01><00><00><1D><00><FF>
The 0x03 "read holding registers" function, asking to read 0x0A=10d registers:
<03><00><00><00><0A>
A second header, with transaction id =2:
<00><02><00><00><00><11><FF>
A 0x10 "write multiple registers" function, writing 0x0A=10d bytes:
<10><00><00><00><05><0A><00><00><00><01><00><02><00><03><00><04> 

Perhaps my question is dumb, but when reading:
 http://www.modbus.org/docs/Modbus_Messaging_Implementation_Guide_V1_0b.pdf
They say p23 that:
"Normally, on MODBUS serial line a client must send one request at a time. This means that  the  client  must  wait  for  the  answer  to  the  first  request  before  sending  a  second  request.   On   TCP/MODBUS,   several   requests   can   be   sent   without   waiting   for   a   confirmation to the same server"

So my maybe dumb question is: was this not the intended behaviour according to spec?

karlp, 2020-01-10:
The length in the initial mbap is indeed wrong, but otherwise, this is just two complete mbap+adu back to back.  I guess the better question is, how is this actually any different from putting a valid length field?  It's still just two requests, and both are therefore handled as usual.  It certainly _could_ drop things, but without any concept of access control in modbus anyway, it seems irrelevant and overthinking it.

See also this remark:
```
All MODBUS requests and responses are designed in such a way that the
recipient can verify that a message is finished. For function codes where
the MODBUS PDU has a fixed length, the function code alone is sufficient.
For function codes carrying a variable amount of data in the request or
response, the data field includes a byte count.
```
This would grant that there's no _reason_ to look at the mbap length in the initial request, as the function code already conveys the required information.  And indeed, given that it "correctly" only consumes the read request, it's "working"

The whole concept of "privilege escalation" is wildly overblown here anyway, there's simply _no_ concept of access control.   Even if you are developing some sort of modbus "firewall" or protection device, you still have all the information you need to evaluate both ADUs independently and allow/disallow them.    Choosing to discard both based on a "wrong" length in one of them seems like the sort of thing security people would do, but that doesn't have any impact anywhere.

karlp, 2020-01-10:
(To follow up, I can't think of any function code where the mbap length would ever be _required_ to be correct, the ADUs themselves _always_ hve enough information in them.  This is why I think no-one ever actually looks at the mbap length.)

ezioyan, 2020-01-11:
Thank you for your patience! I have read modbus implementation guide and figured it out. I thought too much on the problem.

closed 2020-01-11 by ezioyan