stephane/libmodbus #515: Not able to find modbus version

phy455, 2020-01-10:
Hello,

I am using libmodbus version 3.1.4.

In this library I am not able to identify which modbus protocol version  is supported..

Could you  please provide modbus protocol version?

Also in latest release which modbus protocol version is supported?

karlp, 2020-01-10:
what versions of modbus protocol are you looking for?  there's "only one"

Ritesh236, 2020-01-13:
@karlp ,

Thanks for your feedback.

We just want to know more information including version information about modbus protocol which is being used.

Ritesh236, 2020-01-13:
@karlp ,

I just want to know like "MODBUS APPLICATION PROTOCOL SPECIFICATIONV1.1b3" or any other as per information provided over Modbus

karlp, 2020-01-13:
that one then.

Ritesh236, 2020-01-16:
@karlp ,

so libmodbus library is following MODBUS APPLICATION PROTOCOL SPECIFICATIONV1.1b3. correct?

So that we can have idea to move further for that.

Please confirm from your end and also if possible then let us know how i can check into libmodbus library

Regards,
Ritesh Prajapati

david-glover, 2020-01-16:
@Ritesh236,

Yes, libmodbus should conform to that specification.
In the documentation, under external documentation, it links to the Modbus
page, http://www.modbus.org/specs.php.  The 1.1b3 is the most recent
document (2012).  Modbus is not like BACnet, which has multiple protocol
revisions that add new features.

If you want to try out the library, I suggest you clone the git and examine
the tests.  There are client/master and server/slave examples that can be
used for either TCP or serial / RS-485.

Regards,
David

Ritesh236, 2020-01-16:
@david-glover ,

Thanks for your feedback and quick response for same.

Yes. We have already taken that library with version which I have mentioned earlier and also validated client/master and server/slave with examples.

So, Thanks again for confirmation about that and will let you know if any further query regarding that

Regards,
Ritesh Prajapati

karlp, 2020-01-16:
@Ritesh236 can we close this issue then?

Ritesh236, 2020-01-16:
@karlp ,

Yes. We can close this issue now. Thanks for your kind support for that.

I will raise new issue if any related to this.

Regards,
Ritesh Prajapati