stephane/libmodbus #521: How to generate shared library (*.so file) in Linux

hanusek, 2020-02-20:
When:
> ./configure --with-pic --disable-static --enable-shared=yes 
then:
> make --no-print-directory all-recursive
> Making all in src
>   CC       modbus.lo
>   CC       modbus-data.lo
>   CC       modbus-rtu.lo
>   CC       modbus-tcp.lo
>   CCLD     libmodbus.la
> Making all in doc
> make[2]: Nie ma nic do zrobienia w 'all'.
> Making all in tests
> make  all-am
>   CC       bandwidth-server-one.o
>   CCLD     bandwidth-server-one
>   CC       bandwidth-server-many-up.o
>   CCLD     bandwidth-server-many-up
>   CC       bandwidth-client.o
>   CCLD     bandwidth-client
>   CC       random-test-server.o
> random-test-server.c: W funkcji ‘main’:
> random-test-server.c:34:13: warning: implicit declaration of function ‘strcmp’ [-Wimplicit-function-declaration]
>          if (strcmp(argv[1], "tcp") == 0) {
>              ^~~~~~
> random-test-server.c:78:17: warning: declaration of ‘query’ shadows a previous local [-Wshadow]
>          uint8_t query[MODBUS_TCP_MAX_ADU_LENGTH];
>                  ^~~~~
> random-test-server.c:30:12: note: shadowed declaration is here
>    uint8_t *query;
>             ^~~~~
> random-test-server.c:79:13: warning: declaration of ‘rc’ shadows a previous local [-Wshadow]
>          int rc;
>              ^~
> random-test-server.c:27:7: note: shadowed declaration is here
>    int rc;
>        ^~
> random-test-server.c:31:7: warning: variable ‘header_length’ set but not used [-Wunused-but-set-variable]
>    int header_length;
>        ^~~~~~~~~~~~~
> random-test-server.c:30:12: warning: variable ‘query’ set but not used [-Wunused-but-set-variable]
>    uint8_t *query;
>             ^~~~~
> random-test-server.c:28:7: warning: unused variable ‘i’ [-Wunused-variable]
>    int i;
>        ^
> random-test-server.c:27:7: warning: unused variable ‘rc’ [-Wunused-variable]
>    int rc;
>        ^~
>   CCLD     random-test-server
>   CC       random-test-client.o
>   CCLD     random-test-client
>   CC       unit-test-server.o
>

karlp, 2020-02-20:
then do make install.  (or look inside the libtool directories, use ls -la to find them)

hanusek, 2020-02-21:
> then do make install. (or look inside the libtool directories, use ls -la to find them)

I don't want to install. I just want to generate .so files

karlp, 2020-02-21:
So, is this ticket solved then? Can we close it?

closed 2020-02-24 by hanusek