stephane/libmodbus #525: transaction identifier wrong when connection drops, for some reasons I can't understand

LocutusOfBorg, 2020-03-18:
Hello,
I found a really interesting bug in libmodbus, but I don't have yet a clear picture of what is happening.
(modbus version 3.1.6)

I'm using a simple test client that reads registers, and a simple test server that answer to the calls:
```
gcc tests/random-test-server.c -o tests/random-test-server -I src -L ./src/.libs/ -lmodbus
LD_LIBRARY_PATH=./src/.libs ./tests/random-test-server
gcc tests/random-test-client.c -o tests/random-test-client -I src -L ./src/.libs/ -lmodbus
LD_LIBRARY_PATH=./src/.libs ./tests/random-test-client
```

after a while, I just drop the traffic with iptables, and then restore it back
```
iptables -t filter -A INPUT  -i lo -j DROP && sleep 5 &&  iptables -t filter -D INPUT  -i lo -j DROP
```

after the connection is restored, the transaction IDs are wrongly read, and this situation doesn't come back to normality
```
[00][26][00][00][00][06][0D][03][00][25][00][01]
Waiting for a confirmation...
<00><26><00><00><00><05><0D><03><02><00><00>
OK modbus_read_registers single (1)
[00][27][00][00][00][06][0D][03][00][26][00][01]
Waiting for a confirmation...
<00><27><00><00><00><05><0D><03><02><00><00>
OK modbus_read_registers single (1)
[00][28][00][00][00][06][0D][03][00][27][00][01]
Waiting for a confirmation...
ERROR Connection timed out: select
ERROR modbus_read_registers single (-1)
Address = 39
[00][29][00][00][00][06][0D][03][00][24][00][01]
Waiting for a confirmation...
ERROR Connection timed out: select
ERROR modbus_read_registers single (-1)
Address = 36
[00][2A][00][00][00][06][0D][03][00][25][00][01]
Waiting for a confirmation...
ERROR Connection timed out: select
ERROR modbus_read_registers single (-1)
Address = 37
[00][2B][00][00][00][06][0D][03][00][26][00][01]
Waiting for a confirmation...
ERROR Connection timed out: select
ERROR modbus_read_registers single (-1)
Address = 38
[00][2C][00][00][00][06][0D][03][00][27][00][01]
Waiting for a confirmation...
ERROR Connection timed out: select
ERROR modbus_read_registers single (-1)
Address = 39
[00][2D][00][00][00][06][0D][03][00][24][00][01]
Waiting for a confirmation...
<00><28><00><00><00><05><0D><03><02><00><00>
Invalid transaction ID received 0x28 (not 0x2D)
ERROR modbus_read_registers single (-1)
Address = 36
[00][2E][00][00][00][06][0D][03][00][25][00][01]
Waiting for a confirmation...
<00><29><00><00><00><05><0D><03><02><00><00>
Invalid transaction ID received 0x29 (not 0x2E)
ERROR modbus_read_registers single (-1)
Address = 37
[00][2F][00][00][00][06][0D][03][00][26][00][01]
Waiting for a confirmation...
<00><2A><00><00><00><05><0D><03><02><00><00>
Invalid transaction ID received 0x2A (not 0x2F)
ERROR modbus_read_registers single (-1)
Address = 38
[00][30][00][00][00][06][0D][03][00][27][00][01]
Waiting for a confirmation...
<00><2B><00><00><00><05><0D><03><02><00><00>
Invalid transaction ID received 0x2B (not 0x30)
ERROR modbus_read_registers single (-1)
Address = 39
[00][31][00][00][00][06][0D][03][00][24][00][01]
Waiting for a confirmation...
<00><2C><00><00><00><05><0D><03><02><00><00>
Invalid transaction ID received 0x2C (not 0x31)
ERROR modbus_read_registers single (-1)
Address = 36
[00][32][00][00][00][06][0D][03][00][25][00][01]
Waiting for a confirmation...
<00><2D><00><00><00><05><0D><03><02><00><00>
Invalid transaction ID received 0x2D (not 0x32)
ERROR modbus_read_registers single (-1)
Address = 37
[00][33][00][00][00][06][0D][03][00][26][00][01]
Waiting for a confirmation...
<00><2E><00><00><00><05><0D><03><02><00><00>
Invalid transaction ID received 0x2E (not 0x33)
ERROR modbus_read_registers single (-1)
```

the difference in transaction IDs is exactly the number of failed select due to connection drop
Invalid transaction ID received 0x28 (not 0x2D)

cat tests/random-test-server.c
```
/*
 * Copyright © 2008-2014 Stéphane Raimbault <stephane.raimbault@gmail.com>
 *
 * SPDX-License-Identifier: BSD-3-Clause
 */

#include <stdio.h>
#ifndef _MSC_VER
#include <unistd.h>
#endif
#include <stdlib.h>
#include <errno.h>

#include <modbus.h>

int main(void)
{
    int s = -1;
    modbus_t *ctx;
    modbus_mapping_t *mb_mapping;

    ctx = modbus_new_tcp("127.0.0.1", 1502);
    modbus_set_debug(ctx, TRUE);

    mb_mapping = modbus_mapping_new(500, 500, 500, 500);
    if (mb_mapping == NULL) {
        fprintf(stderr, "Failed to allocate the mapping: %s\n",
                modbus_strerror(errno));
        modbus_free(ctx);
        return -1;
    }

    s = modbus_tcp_listen(ctx, 1);
    modbus_tcp_accept(ctx, &s);

    for (;;) {
        uint8_t query[MODBUS_TCP_MAX_ADU_LENGTH];
        int rc;

        rc = modbus_receive(ctx, query);
        if (rc > 0) {
            /* rc is the query size */
            modbus_reply(ctx, query, rc, mb_mapping);
        } else if (rc == -1) {
            /* Connection closed by the client or error */
            break;
        }
    }

    printf("Quit the loop: %s\n", modbus_strerror(errno));

    if (s != -1) {
        close(s);
    }
    modbus_mapping_free(mb_mapping);
    modbus_close(ctx);
    modbus_free(ctx);

    return 0;
}
```

cat tests/random-test-client.c
```
/*
 * Copyright © 2008-2014 Stéphane Raimbault <stephane.raimbault@gmail.com>
 *
 * SPDX-License-Identifier: BSD-3-Clause
 */

#include <stdio.h>
#ifndef _MSC_VER
#include <unistd.h>
#endif
#include <string.h>
#include <stdlib.h>
#include <errno.h>

#include <modbus.h>

/* The goal of this program is to check all major functions of
   libmodbus:
   - write_coil
   - read_bits
   - write_coils
   - write_register
   - read_registers
   - write_registers
   - read_registers

   All these functions are called with random values on a address
   range defined by the following defines.
*/
#define ADDRESS_START    36
#define ADDRESS_END      40
#define SERVER_ID 13

/* At each loop, the program works in the range ADDRESS_START to
 * ADDRESS_END then ADDRESS_START + 1 to ADDRESS_END and so on.
 */
int main(void)
{
    modbus_t *ctx;
    int rc;
    int nb_fail;
    int nb_loop;
    int addr;
    int nb;
    uint16_t *tab_rp_registers;

    /* TCP */
    ctx = modbus_new_tcp("127.0.0.1", 1502);
    modbus_set_debug(ctx, TRUE);
    modbus_set_slave(ctx, SERVER_ID);

    if (modbus_connect(ctx) == -1) {
        fprintf(stderr, "Connection failed: %s\n",
                modbus_strerror(errno));
        modbus_free(ctx);
        return -1;
    }

    /* Allocate and initialize the different memory spaces */
    nb = ADDRESS_END - ADDRESS_START;

    tab_rp_registers = (uint16_t *) malloc(nb * sizeof(uint16_t));
    memset(tab_rp_registers, 0, nb * sizeof(uint16_t));

    for(int i=0; i<1000;i++)
        for (addr = ADDRESS_START; addr < ADDRESS_END; addr++) {
            int i;

            nb = ADDRESS_END - addr;

                rc = modbus_read_registers(ctx, addr, 1, tab_rp_registers);
                if (rc != 1) {
                    printf("ERROR modbus_read_registers single (%d)\n", rc);
                    printf("Address = %d\n", addr);
                    nb_fail++;
                } else {
                    printf("OK modbus_read_registers single (%d)\n", rc);
                }
		sleep(1);
        }

    /* Free the memory */
    free(tab_rp_registers);

    /* Close the connection */
    modbus_close(ctx);
    modbus_free(ctx);

    return 0;
}
```

of course with wireshark, everything is normally sent.
I'm attaching a wireshark trace that shows my server answering correctly.
[modbus.zip](https://github.com/stephane/libmodbus/files/4348812/modbus.zip)

karlp, 2020-03-18:
This is expected as far as I can tell, given the simplistic test code.  The tcp input buffers aren't drained anywhere, so some packets are simply missing.  And the (simplistic) test code never tries to error correct, it just continues expecting that it should live in a perfect world.  If you get unexpected input data, you almost definitely want to use one of the error rrecovery options: https://libmodbus.org/docs/v3.1.6/modbus_set_error_recovery.html

LocutusOfBorg, 2020-03-18:
@karlp looks like even after adding
```
    modbus_set_error_recovery(ctx,
                          MODBUS_ERROR_RECOVERY_LINK |
                          MODBUS_ERROR_RECOVERY_PROTOCOL);
    modbus_flush(ctx);
```

nothing changes, 
```
Waiting for a confirmation...
<00><04><00><00><00><03><0D><84><0B>
Invalid transaction ID received 0x4 (not 0x5)
```

I don't know how, but some bytes are not correctly read from the socket

LocutusOfBorg, 2020-03-18:
For some reasons, it is working now, thanks!
set_error_recovery helped!

closed 2020-03-18 by LocutusOfBorg