stephane/libmodbus #527: *** buffer overflow detected ***: ./unit-test-server terminated

vickymadeshwaran, 2020-03-20:
Please read the following carefully before submitting this new issue.

 - Please ensure, that you are really reporting a bug. When in doubt,
   post a message on https://groups.google.com/forum/#!forum/libmodbus
   or send an email to libmodbus@googlegroups.com

 - Please do not open issues to ask questions about using libmodbus.
   Use the mailing list for this as there are many more people reading
   that list, who could help you.

 - When using libmodbus from a distribution (Debian, Fedora...), please
   report the bug first in the bug tracker of the distribution. The
   reason for doing so is that the package maintainer should have a chance
   to look at the issue first as it might be a packaging error. If/when
   the package maintainer comes to the conclusion that is really an upstream
   bug, then he/she will usually report it here by himself/herself.
   This is because he/she is interested in staying in the notification chain
   to decide about a backport as soon as a bugfix is available.
   Otherwise you (distribution user) will be asked to do so explicitely.

When you get here and you are still convinced that you want report a bug:

 - *Use a clear and decriptive title* for the issue to identify

 - *Which version of libmodbus are you using?* you can obtain this information
   from your package manager or by running `pkg-config --modversion libmodbus`.
   You can provide the sha1 of the commit if you have fetched the code with `git`.

 - *Which operating system are you using?*

 - *Describe the exact steps which reproduce the problem* in as many details as
   possible. For example, the software/equipement which runs the Modbus server, how
   the clients are connected (TCP, RTU, ASCII) and the source code you are using.

 - *Enable the debug mode*, libmodbus provides a function to display the content
   of the Modbus messages and it's very convenient to analyze issues
   (http://libmodbus.org/docs/latest/modbus_set_debug.html).

Good bug reports provide right and quick fixes!

Finally, thank you very much for using libmodbus and taking the time to
file a good bug report. Doing so signals your respect for the developers.

The following template helps you to address the points above. Please delete
everything up to and including the following line which starts with ---.

---

## libmodbus version

  <libmodbus version or sha1 of the latest commit>

## OS and/or distribution

  <e.g. Windows, Linux... version>

## Environment

  <e.g. CPU architecture, 32 vs. 64 bit...>

## Description

  <...>

## Expected behaviour

  <...>

## Actual behaviour

  <...>

## Steps to reproduce the behavior (commands or source code)

  <...>

## libmodbus output with debug mode enabled

  <...>

JoelStienlet, 2020-03-21:
Hello,
Sadly this bug report doesn't provide much information, we have the form without the answers. 
I haven't been able to reproduce this on fedora 31.

ladmanj, 2020-04-01:
Hi
I can't understand what it means here above. Why someone copied a bug report manual here?

Anyway:
After todays kernel update on my ubuntu 19.10 the libmodbus test crashes in this way.
I'm not sure that there is anything more than correlation between the update and the crash.
~~~
libmodbus-master$ tests/bandwidth-server-one rtu
*** buffer overflow detected ***: /home/ladmanj/DataDisk/modbus/libmodbus-master/tests/.libs/bandwidth-server-one terminated
Aborted (core dumped)
~~~

libmodbus installed from sources from this github repo
~~~
$ pkg-config --modversion libmodbus
3.1.6
~~~
~~~
$ uname -a
Linux thinkpad 5.3.0-45-generic #37-Ubuntu SMP Thu Mar 26 20:41:27 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
~~~
Thank you

ladmanj, 2020-04-01:
Oh, i have found it.
The serial port involved disappeared from the system (the USB to serial dongle seems died overnight).

So in my case, the bug is, that libmodbus can open non-existing device and then crashes while preparing handlers for select().

J.
~~~
Breakpoint 1, _modbus_receive_msg (ctx=0x5555555592a0, msg=0x7fffffffda50 "", msg_type=msg_type@entry=MSG_INDICATION)
    at modbus.c:341
341     {
(gdb) list
336        - ETIMEDOUT
337        - read() or recv() error codes
338     */
339
340     int _modbus_receive_msg(modbus_t *ctx, uint8_t *msg, msg_type_t msg_type)
341     {
342         int rc;
343         fd_set rset;
344         struct timeval tv;
345         struct timeval *p_tv;
(gdb) n
350         if (ctx->debug) {
(gdb) 
359         FD_ZERO(&rset);
(gdb) 
360         FD_SET(ctx->s, &rset);
(gdb) 
*** buffer overflow detected ***: /home/ladmanj/DataDisk/modbus/libmodbus-master/tests/.libs/bandwidth-server-one terminated

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb)
~~~

tangxiaozhu666, 2021-05-31:
how to solve this problem？ I just ran into this problem，thanku

nbdy, 2023-01-23:
Same issue here with example below (tried v3.1.6 and v3.1.10).

```cpp
#include <iostream>

#include <modbus/modbus.h>

int main()
{
    modbus_t *ctx = modbus_new_rtu("/dev/ttyUSB1", 19200, 'E', 8, 1);
    modbus_mapping_t *mapping = modbus_mapping_new(500, 500, 500, 500);

    if (mapping == nullptr) {
        std::cout << "Could not allocate mapping: " << modbus_strerror(errno) << std::endl;
        modbus_free(ctx);
        return -1;
    }

    modbus_set_debug(ctx, TRUE);
    std::cout << "Going to run" << std::endl;

    uint8_t query[MODBUS_RTU_MAX_ADU_LENGTH];

    for(;;) {
        int rc = modbus_receive(ctx, query);
        if(rc > 0) {
            modbus_reply(ctx, query, rc, mapping);
        } else if(rc == -1) {
            break;
        }
    }

    std::cout << "Exiting.." << std::endl;

    modbus_mapping_free(mapping);
    modbus_close(ctx);
    modbus_free(ctx);

    return 0;
}
```

The above example produces this output:
```
Going to run
Waiting for an indication...
*** buffer overflow detected ***: terminated
```

As @ladmanj already mentioned [above](https://github.com/stephane/libmodbus/issues/527#issuecomment-607125504), this stems from the _modbus_receive_msg function.
I do not have an issue when using the [QModbusRtuSerialServer](https://doc.qt.io/qt-6/qmodbusrtuserialserver.html) or [umodbus](https://umodbus.readthedocs.io/en/latest/).
Also I've tried multiple USB-RS232 adapters.

stephane, 2024-10-22:
`./unit-test-server` is designed for Unit tests. It won't try to recover failing communication.

closed 2024-10-22 by stephane