stephane/libmodbus #531: Indication Not Sent After Confirmation Is Received

napperley, 2020-04-11:
## libmodbus version

Version 3.1.6 compiled from the latest stable release from the GitHub repository.

## OS and/or distribution

- OS: Linux
- Distribution: Linux Mint 19.3

## Environment

- **CPU:** AMDx64
- **Network:** Loopback device

## Description

When running both the Modbus TCP client, and Server the client sends a request (function type *04*) to the server. On the server it receives the request, and sends a confirmation to the client. After the client receives the confirmation it prints out the values of the input registers, however it doesn't send back a indication to the server. Eventually a indication timeout occurs which causes the server to exit gracefully.

## Expected behaviour

Client sends back a indication **after** receiving a confirmation from the server.

## Actual behaviour

No indication is sent from the client **after** the client receives a confirmation from the server.

## Steps to reproduce the behaviour (commands or source code)

1. Clone the Modbus Client [repository](https://gitlab.com/napperley/modbus-client)
2. Clone the Modbus Server [repository](https://gitlab.com/napperley/modbus-server)
3. Change working directory to where the Modbus Server was cloned to
4. Build the Modbus Server: `./gradlew linkModbus_serverDebugExecutableLinuxX64`
5. Change working directory to **build/bin/linuxX64/modbus_serverDebugExecutable**
6. Run the server: `./modbus_server.kexe 127.0.0.1 10000`
7. Open a new terminal
8. Change working directory to where the Modbus Client was cloned to
9. Build the Modbus Client: `./gradlew linkModbus_clientDebugExecutableLinuxX64`
10. Change working directory to **build/bin/linuxX64/modbus_clientDebugExecutable**
11. Run the client: `./modbus_client.kexe 0 4 3 127.0.0.1:10000`

## libmodbus output with debug mode enabled

**Server:**
```
The client connection from 127.0.0.1 is accepted
Waiting for an indication...
<00><01><00><00><00><06><03><04><00><00><00><04>
[00][01][00][00][00][0B][03][04][08][01][7F][00][56][01][4D][00][46]
Waiting for an indication...
ERROR Connection reset by peer: read
```

**Client:**
```
Connecting to 127.0.0.1:10000
[00][01][00][00][00][06][03][04][00][00][00][04]
Waiting for a confirmation...
<00><01><00><00><00><0B><03><04><08><01><7F><00><56><01><4D><00><46>
```

karlp, 2020-04-12:
What sort of "confirmation" are you expecting the client to send?  The second "waiting for indication" is exactly that, the server waiting for a new request, and when the client exits, the socket closes.  This is just a problem with your event loop runner.  You do accept once, but then you do a while(true) loop on receiving request, and the second request will never arrive.  I don't see any problem here.

napperley, 2020-04-13:
Fixed the issue with the server implementation, which will work fine for a single connection. With multiple connections this will require using coroutines. In the coroutines version the main function blocks for connections, and the runEventLoop function (for handling register read events) is non blocking (a coroutine), along with the use of the Actor concurrency model (via a separate coroutine) to manage the connection counter.

closed 2020-04-13 by napperley