stephane/libmodbus #535: modbus_read_registers

viktor-morin, 2020-04-23:
I'm running following code on Linux:
     
    
    #include <stdio.h>
	#include <modbus.h>

	int main(void)
	{
	  modbus_t *ctx;
	  uint16_t tab_reg[64];
	  int rc;
	  int i;

	  ctx = modbus_new_tcp("127.0.0.1", 502);
	  if (modbus_connect(ctx) == -1)
	  {
		printf("Connection failed: %d \n", -1);
		modbus_free(ctx);
		return -1;
	  }

	  rc = modbus_read_registers(ctx, 20000, 10, tab_reg);
	  if (rc == -1)
	  {
		printf("Error reading register %d \n", rc);
		return -1;
	  }

	  for (i = 0; i < rc; i++)
	  {
		printf("reg[%d]=%d (0x%X)\n", i, tab_reg[i], tab_reg[i]);
	  }

	  modbus_close(ctx);
	  modbus_free(ctx);
	}`

Result:

> Error reading register -1

I want to read from register 20000
Using a C# client called EasyModbus Client (Free Software), I get it to work:
[https://imgur.com/zQfxd28](https://imgur.com/zQfxd28)

What's wrong?

karlp, 2020-04-24:
try using the recommended code from the documentation to get better errors.  You might also like to turn on verbose debug to see what's sent/received and compare.
```
rc = modbus_read_registers(ctx, 20000, 10, tab_reg);
if (rc == -1) {
    fprintf(stderr, "%s\n", modbus_strerror(errno));
    return -1;
}
```

viktor-morin, 2020-04-25:
using that example, errno isnt defined.

karlp, 2020-04-28:
then include errno.h, errno is a global...

viktor-morin, 2020-04-29:
Updated the code to follwing:
```rc = modbus_read_registers(ctx, 20000, 10, tab_reg);
if (rc == -1)
{
	printf("Error reading register %d \n", rc);
	fprintf(stderr, "%s\n", modbus_strerror(errno));
	return -1;
}
```
Get following output:

> Error reading register -1
Connection timed out

Again, it works with other clients like [https://imgur.com/zQfxd28](https://imgur.com/zQfxd28)

mhei, 2020-04-29:
Repeating that a C# client works for this Modbus Server does not help very much. I doubt that this C# client runs in the same environment (Linux) as your libmodbus-based program, does it?

So now that you have a human-readable error message for the reason, check libmodbus documentation how you could play with libmodbus' API to address this issue, e.g. enable debugging, modify timeouts...
I'd also recommend to use Wireshark to inspect the TCP data flow...

karlp, 2020-04-29:
(is your target device actually running on 127.0.0.1?)

viktor-morin, 2020-04-30:
I was running on Linux WSL (Windows Subsystem Linux), but to make sure that wasnt an issue I connected a Raspberry Pi. I still get same error:

> Error reading register -1
Connection timed out

And I have verified it works from other clients, and that I can ping the device from Linux.
So its clear to me that there is something, probably a settings or similar I have to change?
But usually when you follow an example provided by the lib itself, they tend to work without additional configuration. 

I will see if I can find anything in the documentation regarding this

viktor-morin, 2020-04-30:
Uploaded picture from WireShark: [https://imgur.com/tOWaWcd](https://imgur.com/tOWaWcd),
and the log itself: [https://www.dropbox.com/s/awmnho9da9b0d7c/modbus.pcapng?dl=0](https://www.dropbox.com/s/awmnho9da9b0d7c/modbus.pcapng?dl=0)

viktor-morin, 2020-05-19:
Solved my problem by using another lib

closed 2020-05-19 by viktor-morin

JoelStienlet, 2020-05-22:
`ctx = modbus_new_tcp("127.0.0.1", 502);`
here the library is asked to open a TCP connection to locahost. But the wireshark screenshot (log link broken) shows two machines:
- 192.168.2.100
- 192.168.2.11
Has the wireshark log been taken with libmodbus, or with the other "EasyModbus C# client"?
It's not clear what went wrong here.

netomx, 2021-03-01:
What library did you use? I have some issues that I need to try to send some registers, and I need to send them several times.