stephane/libmodbus #537: libmodbus没有关于W5500以太网接口的说明

zhigangbox, 2020-04-26:
平台：rt_thread +stm32f413+w5500+libmodbus
错误描述：
       我是用w5500来实现modbus TCP（W5500已经可以PING通）,但是libmodbus关于不同以太网接口的部分并没有仔细说明，我将ctx = modbus_new_tcp(RT_NULL, 1502, AF_WIZ);中的AF_INET  改成了AF_WIZ，其它地方没有再有说明，运行后终端输出“"WIZnet socket (%d) not support option name”。
问题：
      1 我用W5500实现modbus TCP该怎么移植？
      2  modbus_new_tcp中为何是 RT_NULL，而不是W5500自己设置的IP地址
      3 动态分配的寄存器而非静态数组形式在KEIL MDK中如何查看修改了的数据？

JoelStienlet, 2020-04-26:
Hello, 
I copy-paste the translation so that others don't have to do it:
Platform: rt_thread + stm32f413 + w5500 + libmodbus
Error description:
I used w5500 to implement modbus TCP (W5500 can be pinged), but the part of libmodbus about different Ethernet interfaces is not carefully explained, I will ctx = modbus_new_tcp (RT_NULL, 1502, AF_WIZ); AF_INET is changed to AF_WIZ, there is no description in other places, the terminal outputs "" WIZnet socket (% d) not support option name "after running.
Questions:
1 How do I use W5500 to implement modbus TCP how to transplant ?
2 modbus_new_tcp in what is RT_NULL, rather than the IP address of W5500 own set of
3 dynamic rather than static allocation of an array of how to register see the modified data in KEIL MDK in?

My 2 cent: I wonder if you can use libmodbus on this embedded system, it depends on what the "rt_thread" OS provides, and maybe you can port the library to it. From the limodbus website, the officially supported OSes are: "Linux, Mac OS X, FreeBSD, QNX and Win32".
Normally, your w5500 should be managed by the kernel, not libmodbus, which means, either "rt_thread" has a built-in driver for your w5500, or you should add it to the rt_thread kernel yourself. Note that rt_thread's kernel also has to manage the TCP/IP stack. The TCP/IP layer is not managed by libmodbus, which only focuses on the modbus protocol.
I hope this helps.

zhigangbox, 2020-04-27:
在RT_THREAD我也没找到帮助。
我想说的是，modbus TCP是基于tcp server的，libmodbus作为应用层协议，如果用w5500的tcp server收到连接客户端信息后，将其交给modbus解析（这应该是freemodbus的架构）。
但是libmodbus的架构是创建服务器，侦听，接收回应，相当于直接集成了tcp server的驱动，也给出了AF_INET（一般TCP/IP） ,AF_AT（2G/3G/4G基于AT命令集）,AF_WIZ（类似W5500集成硬件TCP/IP协议栈）三种类型，那么就应该对基于AF_WIZ的W5500给出接口操作及说明，不然也没法操作啊。

备注：基于rt_thread的libmodbus(本源码实现了在RT-Thread上的移植，支持MODBUS-RTU和MODBUS-TCP。源码地址：) :
https://github.com/loogg/libmodbus 作者：loogg.

JoelStienlet, 2020-04-27:
Hello,
> libmodbus作为 应用层协议

yes, this library is at the application layer only (layer 5), it is not aware of any hardware layer details. 
This library interfaces with the kernel using the connect() system call:
[man page of connect](http://man7.org/linux/man-pages/man2/connect.2.html)
[list of Linuxe's system calls](http://man7.org/linux/man-pages/man2/syscalls.2.html)
It expects the kernel to deal with the transport, network, datalink and physical interface (TCP/IP network stack + harware drivers).
I quickly looked at this RT-Thread, and apparently they provide some features to deal with your w5500, and present a connect() interface to the userspace (where this library resides):
[RT-thread socket abstraction layer](https://github.com/RT-Thread/rtthread-manual-doc/blob/master/sal/sal.md)
You can try to compile (or cross compile) this library for RT-Thread using this SAL layer, hopefully you don't hit any major issues.

stephane, 2020-06-11:
Thank you @JoelStienlet

closed 2020-06-11 by stephane