stephane/libmodbus #544: Get rid of master/slave terms

stephane, 2020-06-16:
The terms master/slave come from the Modbus protocol documentation.

A few years ago, I considered replacing them by server/client, to avoid racist connotations and improve understanding.

Today, the time has come to consider this change as urgent.
✊🏿✊🏽✊🏻

watsocd, 2020-06-16:
I understand where you are coming from but I think we need different terms than client/server.

 

The word server implies more power than a MODBUS slave has to offer.

 

The work client implies less power than a MODBUS master has over the network. 

 

I Googled “alternatives to master slave terminology” but many terms were not relevant to what we do with a poll/reply protocol like MODBUS. 

 

Other terms were more about master and backup scenarios like with databases.

dreamflow, 2020-06-23:
> I considered replacing them by server/client

@stephane :

i'm also in favour for using the words `client/server` because :

1 . often times client/server is used for descriptions of
modbus , so most modbus users are probably already
accustomed to the words `client/server` .

2. the modbus-protocol is a client/server-protocol
in the sense/definition of computer science  .

 > The word server implies more power than a MODBUS slave has to offer.

@watsocd :

i disagree . for a server it doesn't matter how big or small it is or
how many or few features it has . what matters is the service it serves ,
even if it's just about 1 register for 1 temperature-measurement-value
from 1 temperature-sensor . or an even smaller use case with just
1 modbus coil , to control an on/off state of a LED for example .
even such smallest possible examples are still modbus in a
client/server-scenario .

xinthose, 2020-09-09:
I guess in a restaurant the slave is the server and the master is the client, kind of sounds counter intuitive.

watsocd, 2020-09-09:
To use your restaurant analogy, the counter is the slave interface (RS485/TCP) and the people behind the counter are the worker/slave threads in the remote device.

 

The customer/master makes a request at the counter for certain products/registers. 

 

The workers make/retrieve the food/data and serve the food/data over the counter/interface or announce product/register not available/NACK.

 

Nothing crosses the counter/interface until a customer makes a request.

karlp, 2020-09-10:
I'm not a big fan of server/client as replacements as it muddies the water on which is the server.  Is the old slave the server?  It is in a pure CS sense, and some modbus tcp docs talk like that, but in the sense of the server running the show, as most users expect, that's the modbus master, which is the client.  I think modbus needs different terms than just blindly switching to client/server.  (I'm not even sure which way around you were proposing)

dreamflow, 2020-10-01:
> Is the old slave the server?

yes .
a typical example is a modbus-capable measurement-device(e.g. for temperature ,
voltage , current , ...) , such a device is called `server` in `client/server` wording .

here is a list of some modbus-libraries that are using the `client/server` wording :

python :
- pymodbus : [homepage](https://riptideio.github.io/pymodbus) , [github](https://github.com/riptideio/pymodbus)
- µModbus : [homepage](https://umodbus.readthedocs.io) , [github](https://github.com/AdvancedClimateSystems/uModbus)
- EasyModbusTCP.PY : [github](https://github.com/rossmann-engineering/EasyModbusTCP.PY)

JavaScript / Node.js :
- https://github.com/yaacov/node-modbus-serial
- https://github.com/Cloud-Automation/node-modbus
- https://github.com/dresende/node-modbus-tcp
- https://github.com/node-modbus/stream

ruby :
- RModBus : [homepage](https://www.rubydoc.info/github/rmodbus/rmodbus) , [github](https://github.com/rmodbus/rmodbus)

.NET :
- FluentModbus : [homepage](https://apollo3zehn.github.io/FluentModbus) , [github](https://github.com/Apollo3zehn/FluentModbus)
- https://github.com/rossmann-engineering/EasyModbusTCP.NET
- https://github.com/nickvdyck/libmodbus.cs
- https://github.com/bradder555/fsModbus

rust :
- rmodbus : [homepage](https://crates.io/crates/rmodbus) , [github](https://github.com/alttch/rmodbus)

GDV0, 2021-01-28:
It is a very old and passionate debate!

Several device typologies exist in networking:
- **Master/slave** : A Master is a device which manages all the exchanges on the network allowing other devices (Slaves) to send and receive information (e. g. I2c, is a master/slave protocol where Master is generating a clock for the Slaves to send data)
- **Client/Server**: A Client is a device wich requests information from other devices (Servers)
- **Provider/Consumer** : A Provider is a device sending data on the network without having to receive external request for them, just expecting these data wil be picked up by Consumers to do something with. ( CAN  is a Provider/Consumer protocol)
- **Publisher/Subscriber**: derived from Provider/Consumer, the Publisher will wait for a Data subscription from other device (Subscriber) before acting as a Provider for this data while Subscriber will act as a Consumer when seeing the data on the network.
- ...

Confusion between Master/Slave and Client/Server for MODBUS is due to the fact that MODBUS Serial allows only 1 device to send requests to other devices: 
- this device acts as Client, and others devices as Servers following the definitions I gave previously. 
- If this device is not connected or not sending requests, there is no exchanges. meaning that this device could be considered as a Master and other devices as Slaves.

The right answer will be probably given by Modbus.org which tries to remove Master and Slave words from their MODBUS Serial documents (MODBUS/TCP is already and clearly identified as a Client/Server protocol). 
Unfortunately, as this is an old protocol (more than 40 years), specification updates are not so frequent.

jaskij, 2021-02-26:
Similarly, the application protocol spec is also changed (there's one "master/slave" pair in an image which likely is an omission)

Links:
- [declaration of name change](https://modbus.org/docs/Client-ServerPR-07-2020-final.docx.pdf]
- [current application protocol spec](https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf)

Ekristoffe, 2022-04-14:
> The terms master/slave come from the Modbus protocol documentation.
> 
> A few years ago, I considered replacing them by server/client, to avoid racist connotations and improve understanding.
> 
> Today, the time has come to consider this change as urgent. ✊🏿✊🏽✊🏻

Master/Slave is the right terminology for the communication case because there is only one master allowed for multiple slaves.
Master/Slave in our world should be only used for those and should not be "cancelled" because some "snowflake" want to force their "wokeness" point of view on the world. I am sorry but I am against this stupid politic of "politically correct" just because of "woke" culture. This is another reason why the "West" is so laugh about in Asia and other part of the world, only doing what the louder minority want and crying racism for everything which is against their agenda.

Server/Client is another term which meaning is different. One Server can serve more than one Client.
Master/Slave put the emphasis on the ownership on the communication.
This is one reason why TCP/UDP communication can be multi client and not the RTU/ASCII one.

Guiorgy, 2022-06-08:
Firstly, Master/Slave is just a naming convention. I can't imagine anybody being offended when we are referring to inanimate objects.

Secondly, as many pointed out, Client/Server doesn't quite fit in our context, at least not in case of RTU.

And thirdly, unless everybody using Modbus agrees to change the naming convention to something else, I hope this library doesn't do that, since it would only cause needless confusion.

>     1. often times client/server is used for descriptions of
>        modbus , so most modbus users are probably already
>        accustomed to the words `client/server` .

No, this would completely confuse my coworkers. All our documentation around our PLC and the software around it uses the Master/Slave convention.

Finally, before we discus anything related to changing the standard naming convention, I'd like to propose to at least change the main branch of the repository from `master` to `main`, as is the preferred way nowadays. :)

stephane, 2022-10-03:
It was fun to piss off the fascists but the implementation is not realistic.
It will just try to rename the main branch.

closed 2022-10-03 by stephane