stephane/libmodbus #547: MODBUS_GET_INT64_FROM_INT16 registers' order is wrong.

jajosheni, 2020-06-23:
## libmodbus version

 3.1.6

## OS and/or distribution

 Ubuntu 18.04 & 16.04

## Environment

 64bit

## Description

MODBUS_GET_INT64_FROM_INT16 does not return the right value.
 https://github.com/stephane/libmodbus/blob/master/src/modbus.h#L247

## Expected behaviour

  3254

## Actual behaviour

  6223372036854775807

## Steps to reproduce the behavior (commands or source code)

  Try to read any INT64 from  INT16 registers

## MY FIX :
```
#define MODBUS_GET_INT64_FROM_INT16(tab_int16, index) \
    (((int64_t)tab_int16[(index) + 3 ] << 48) | \
     ((int64_t)tab_int16[(index) + 2] << 32) | \
     ((int64_t)tab_int16[(index) + 1] << 16) | \
      (int64_t)tab_int16[(index)  ])
```

karlp, 2020-06-24:
_If_ you change it like this, you would need to change the 32bit versions as well.  You're changing the byte ordering, and typically, modbus follows MSB first, which is why it's like it is.  you want a new function for get 32/64_le or similar

jajosheni, 2020-06-24:
> 
> 
> _If_ you change it like this, you would need to change the 32bit versions as well. You're changing the byte ordering, and typically, modbus follows MSB first, which is why it's like it is. you want a new function for get 32/64_le or similar

a new `#define` would be just fine. I am working with PLCs right now and every time I install libmodbus I have to change it. 
Maybe the best would be an option to specify byte order.

karlp, 2020-06-24:
I understand, I have extra wrappers in my own code too, because it's very much "every way goes" and while your PLC needs the switch, it's hardly all of them :) witness for instance: "modbus_get_float_abcd(3) modbus_set_float_abcd(3) modbus_get_float_badc(3) modbus_set_float_badc(3) modbus_get_float_cdab(3) modbus_set_float_cdab(3) modbus_get_float_dcba(3)"

To clarify, I don't believe it's _wrong_ as it stands, I believe you want a new feature to have LSB ordered 32/64bit integer helpers, similar to the LSB/mixed ordered helpers for floats.

jajosheni, 2020-06-25:
I updated the Title and the content.

Maybe in the future someone will add an option to change bytes order.
I hope the owner will put a "Feature" label over this issue.

karlp, 2020-06-26:
You can leave this open though?

jajosheni, 2020-06-30:
sure

stephane, 2022-02-21:
What's the point to leave this issue opens?

jajosheni, 2022-02-22:
@stephane it is a requested feature, but i'm not up to date if it was added.
feel free to close it

stephane, 2022-02-22:
I'm a bit reluctant to add other variants of MODBUS_GET/SET_* (+ tests, docs, etc) because I'm not sure it will be useful or clear on how to use them.
As @karlp, each user tends to create their own wrappers.

So I propose to clear the issue for now and to see if asked again in the future.

closed 2022-02-22 by stephane