stephane/libmodbus #549: Add support for Modbus RTU Over UDP

systemmonkey42, 2020-06-24:
Is there any interest in building UDP support into libmodbus?

I recognise that Modbus/UDP is not (yet?) a recognised standard, however it is becoming not only 
available in modern hardware, but recommended by the vendors for certain workloads.

Given that UDP simply requires the exact same MBAP and payload as TCP,  but over a different transport,
can it really be that difficult?

I'm a little suprised there are not forks with UDP support.  The only opensource UDP implementation I've
found on github are python, node (javascript) or .NET.  None of which are useful to my project.

Thanks

mdejw, 2021-12-14:
I also looking for UDP MODBUS access.

systemmonkey42, 2021-12-15:
I ended up adding UDP support to 3.1.6 library release.  Its been in production for over a year now and seems to work well.

I've merged my patches with master and published them here.

https://github.com/systemmonkey42/libmodbus/tree/udp_support

mdejw, 2021-12-15:
Thank you for info and sharing :)

I'll try it.

systemmonkey42, 2021-12-15:
> Thank you for info and sharing :)
> 
> I'll try it.

No problem.  I included the C code I linked with the customer application (which provides a LUA library).
They run a control module which launches a LUA application which performs the modbus control.

They perform all modbus specific magic in LUA, and this code provides the interface.

PythonWanna, 2023-09-18:
> > Thank you for info and sharing :)
> > I'll try it.
> 
> No problem. I included the C code I linked with the customer application (which provides a LUA library). They run a control module which launches a LUA application which performs the modbus control.
> 
> They perform all modbus specific magic in LUA, and this code provides the interface.
> 
> Edit: Remember this is "modbus tcp/ip" using a UDP socket. Not ModbusRTU over UDP (which I understand might have a different payload format).

Hello, I think you should add this line of code, otherwise the data can't be updated! @systemmonkey42 
[image]