stephane/libmodbus #552: modbus_rtu_set_rts - not pretty valid example in "/doc/modbus_rtu_set_rts.txt"

DimionDR, 2020-07-15:
## libmodbus version

  f1eb4bc7ccb09cd8d19ab641ee37637f8c34d16d

## Description

  Example:
```
EXAMPLE
-------
.Enable the RTS mode with positive polarity
[source,c]
-------------------
modbus_t *ctx;
uint16_t tab_reg[10];

ctx = modbus_new_rtu("/dev/ttyS0", 115200, 'N', 8, 1);
modbus_set_slave(ctx, 1);
modbus_rtu_set_serial_mode(ctx, MODBUS_RTU_RS485);
modbus_rtu_set_rts(ctx, MODBUS_RTU_RTS_UP);

if (modbus_connect(ctx) == -1) {
    fprintf(stderr, "Connexion failed: %s\n", modbus_strerror(errno));
    modbus_free(ctx);
    return -1;
}

```
But in such case function `modbus_rtu_set_serial_mode` will try to access ctx->s before it was opened in `modbus_connect`:
```
int modbus_rtu_set_serial_mode(modbus_t *ctx, int mode)
{
    if (ctx == NULL) {
        errno = EINVAL;
        return -1;
    }

    if (ctx->backend->backend_type == _MODBUS_BACKEND_TYPE_RTU) {
#if HAVE_DECL_TIOCSRS485
        modbus_rtu_t *ctx_rtu = ctx->backend_data;
        struct serial_rs485 rs485conf;

        if (mode == MODBUS_RTU_RS485) {
            // Get
            if (ioctl(ctx->s, TIOCGRS485, &rs485conf) < 0) {
                return -1;
            }
            // Set
            rs485conf.flags |= SER_RS485_ENABLED;
            if (ioctl(ctx->s, TIOCSRS485, &rs485conf) < 0) {
                return -1;
            }

            ctx_rtu->serial_mode = MODBUS_RTU_RS485;
            return 0;
        } else if (mode == MODBUS_RTU_RS232) {
...
```
## Expected behaviour

```
EXAMPLE
-------
.Enable the RTS mode with positive polarity
[source,c]
-------------------
modbus_t *ctx;
uint16_t tab_reg[10];

ctx = modbus_new_rtu("/dev/ttyS0", 115200, 'N', 8, 1);
modbus_set_slave(ctx, 1);

if (modbus_connect(ctx) == -1) {
    fprintf(stderr, "Connexion failed: %s\n", modbus_strerror(errno));
    modbus_free(ctx);
    return -1;
} else {
  modbus_rtu_set_serial_mode(ctx, MODBUS_RTU_RS485);
  modbus_rtu_set_rts(ctx, MODBUS_RTU_RTS_UP);
}

```

robin48gx, 2020-12-16:
I cannot find MODBUS_RTU_RTS_UP in libmodbus 3.0.6 .

karlp, 2020-12-16:
> I cannot find MODBUS_RTU_RTS_UP in libmodbus 3.0.6 .

it's in 3.1.x and up only

closed 2024-10-22 by stephane