stephane/libmodbus #557: build lib and dll with vs2015 under win64

YiGene, 2020-08-26:
it take me some time to get it build to 64bit under 2015 hence i put a note here 

after we choose x64 to debug and release, run build 
[image]

we see below errors:
1>         d:\modbus\libmodbus-3.1.6\src\modbus.h(228): note: see previous definition of 'modbus_mapping_free'
1>         Generating Code...
1>       Link:
1>            Creating library D:\modbus\libmodbus-3.1.6\src\win32\x64\Debug\modbus.lib and object D:\modbus\libmodbus-3.1.6\src\win32\x64\Debug\modbus.exp
1>     1>
1>modbus-rtu.obj : warning LNK4217: locally defined symbol modbus_free imported in function modbus_new_rtu
1>     1>
1>modbus-tcp.obj : warning LNK4217: locally defined symbol modbus_free imported in function _modbus_tcp_init_win32
1>     1>
1>modbus.obj : warning LNK4217: locally defined symbol modbus_set_bits_from_bytes imported in function modbus_reply
1>     1>
1>modbus-data.obj : error LNK2019: unresolved external symbol __imp_htonl referenced in function modbus_set_float_abcd
1>     1>

add ws2_32.lib
[image]

build success

>modbus.obj : warning LNK4217: locally defined symbol modbus_set_bits_from_bytes imported in function modbus_reply
1>         modbus.vcxproj -> D:\modbus\libmodbus-3.1.6\src\win32\x64\Debug\modbus.dll
1>         modbus.vcxproj -> D:\modbus\libmodbus-3.1.6\src\win32\x64\Debug\modbus.pdb (Full PDB)
1>       FinalizeBuildStatus:
1>         Deleting file "x64\Debug\modbus.tlog\unsuccessfulbuild".
1>         Touching "x64\Debug\modbus.tlog\modbus.lastbuildstate".
1>
1>Build succeeded.
1>
1>Time Elapsed 00:00:12.33
========== Build: 1 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========

**PLEASE NOTE i didn't test the function, just build it **

AndoniC, 2020-10-21:
Hi YiGene, 
Relating to this I wrote thissimple CMakelists.txt in order to create Visual Studio solutions and it seems it can also be compiled and it works ok, just if it is of interest.

`cmake_minimum_required(VERSION 3.0 FATAL_ERROR)

#include("./TargetArch.cmake")
#target_architecture (TARGET_COMPILATION_ARCH)
#check : IF(TARGET_COMPILATION_ARCH MATCHES x86_64)
#message(STATUS "** Compilation Target is ${TARGET_COMPILATION_ARCH} bits **")

PROJECT (modbus)

#Version
SET( modbus_VERSION_MAJOR 1 )
SET( modbus_VERSION_MINOR 1 )
SET( modbus_VERSION_PATCH 0 )
SET( modbus_VERSION_STATUS "dev") 
SET( modbus_PROJECT_VERSION ${modbus_VERSION_MAJOR}.${modbus_VERSION_MINOR}.${modbus_VERSION_PATCH}-${modbus_VERSION_STATUS})

#Set folders in Visual studio
set_property( GLOBAL PROPERTY USE_FOLDERS On)

#Source files
FILE (GLOB sources "src/*.h" "src/*.c")

#Create config.h from config.h.win32
CONFIGURE_FILE("${PROJECT_SOURCE_DIR}/src/config.h.win32" "${PROJECT_BINARY_DIR}/config.h" @ONLY)
#Include the directory containing config.h in include directories list
include_directories(${PROJECT_BINARY_DIR})

#add target 
add_library(modbus SHARED ${sources} "${PROJECT_BINARY_DIR}/config.h")

#add ws2_32 library
target_link_libraries(modbus ws2_32 )

#add preprocessor directives 
target_compile_definitions( modbus PUBLIC  "-DDLLBUILD")
target_compile_definitions( modbus PUBLIC  "MODBUS_VERSION=${modbus_PROJECT_VERSION}")
CONFIGURE_FILE("${PROJECT_SOURCE_DIR}/config.cmake.in" "${PROJECT_BINARY_DIR}/modbusConfig.cmake" @ONLY)
`

config.cmake.in is the file used to create modbusConfig.cmake (used by FindPackage to generate configuration variables). Here the content:

`
#  Usage from an external project:
#    In your CMakeLists.txt, add these lines:
#
#       FIND_PACKAGE(modbus REQUIRED )
#	 INCLUDE_DIRECTORIES(${modbus_INCLUDE_DIRS})
#       TARGET_LINK_LIBRARIES(TARGET_NAME ${modbus_LIBRARIES})
#
#    This file will define the following variables:
#      - modbus_LIBRARIES        : The list of libraries to links against.
#      - modbus_INCLUDE_DIRS  : The directory where include files are.
#      - modbus_LIB_DIR       : The directory where lib files are. Calling LINK_DIRECTORIES
#                                with this path is NOT needed.
#      - modbus_VERSION       : The  version of this build. Example: "1.2.0"
#      - modbus_VERSION_MAJOR : Major version part of VERSION. Example: "1"
#      - modbus_VERSION_MINOR : Minor version part of VERSION. Example: "2"
#      - modbus_VERSION_PATCH : Patch version part of VERSION. Example: "0"
set (modbus_INCLUDE_DIRS "@PROJECT_BINARY_DIR@" "@PROJECT_SOURCE_DIR@/src")
set (modbus_LIB_DIR  "@PROJECT_BINARY_DIR@/$(Configuration)")
set(modbus_LIBRARIES debug @PROJECT_BINARY_DIR@/Debug/modbusd.lib optimized @PROJECT_BINARY_DIR@/Release/modbus.lib)
set(modbus_VERSION 		  @modbus_VERSION@)
set(modbus_VERSION_MAJOR  @modbus_VERSION_MAJOR@)
set(modbus_VERSION_MINOR  @modbus_VERSION_MINOR@)
set(modbus_VERSION_PATCH  @modbus_VERSION_PATCH@)
`

stephane, 2022-11-28:
Related to 9a7ea129283a8a69de0f36572ad2f9261d0003ac

msk-repo-0x4d, 2023-04-15:
Hi @stephane and everyone.
I have compiled and used this library in both Windows and Linux. I found that the README file for Windows (inside **src/win32** folder) is not complete as it does not tell the user how to create a **dll** out of this. For creating **dll** I had to add "**___declspec(dllexport)_**" to all the functions that I used in my code. 

I am not sure if you consider this useful but if you do then I am ready to volunteer to make the README file bigger with steps to create a **dll** file.

Just to elaborate I added a Macros **DLL_EXPORT** in ***modbus.h*** file as below:

```
#ifdef _WIN32
    #define DLL_EXPORT __declspec(dllexport)
#endif
```

And then in appropriate header file I added this **DLL_EXPORT** to all the important methods that should be exported. One Example is shown :

```
MODBUS_API modbus_t * DLL_EXPORT
modbus_new_rtu(const char *device, int baud, char parity, int data_bit, int stop_bit);
```

 Let me know what you think of this making README more elaborative!

stephane, 2023-04-17:
Yes please, this would be very useful information.

msk-repo-0x4d, 2023-04-19:
@stephane I just created a pull request (https://github.com/stephane/libmodbus/pull/697) after making changes. Let me know if there is something I have to explain better. Also, I left the Visual Studio part as it is, as I don't have Visual Studio right now. Hope this helps!

msk-repo-0x4d, 2023-05-03:
Hi @stephane. Did you get a chance to look at the new `README` for Windows (under **src/win32**)?

ottelo9, 2023-05-12:
Why no one can leave the build dll files here so that we can download them? Rights problems? Can anyone please upload it? THX

msk-repo-0x4d, 2023-05-16:
@ottelo9 Check the wiki page of this project. They have mentioned a google-drive link to share it. Navigate here in case you are having difficulty:
https://github.com/stephane/libmodbus/wiki/Compile-dll-under-Windows-with-Visual-Studio-2008-Express#ready-to-use-dll

ottelo9, 2023-05-17:
thanks for your answer. That is my dll that I uploaded few years ago 😂😂😂. Yesterday I compiled v3.1.10. I will upload it again and put it on wiki.

luocheng610, 2023-07-13:
> thanks for your answer. That is my dll that I uploaded few years ago 😂😂😂. Yesterday I compiled v3.1.10. I will upload it again and put it on wiki.

It seems that the 64-bit version cannot be compiled. I can compile the 32-bit version. Can you compile the 64-bit version of the DLL?

ottelo9, 2023-09-19:
@luocheng610 
I use Visual Studio 2008 Express. I can only compile win32 versions. I dont know how to compile 64Bit :( . But please! If anyone can share a 64Bit libmodbus, please do it.

luocheng610, 2023-09-20:
> @luocheng610 I use Visual Studio 2008 Express. I can only compile win32 versions. I dont know how to compile 64Bit :( . But please! If anyone can share a 64Bit libmodbus, please do it.

No problem, I'll upload it
[libmodbus-3.1.10_VS2008_X64.zip](https://github.com/stephane/libmodbus/files/12671301/libmodbus-3.1.10_VS2008_X64.zip)

ottelo9, 2023-09-20:
@luocheng610 
Thanks. I put it to the Wiki.

AbhishekBade1310, 2024-06-18:
i want to make a dll with libmodbus and then use that dll in c#. the issue I'm facing is that when i change the "common language runtime" support field in config properties-->advanced so the dll can be called in c#, it throws error saying that the .c files can't be used.

[image]

AbhishekBade1310, 2024-06-19:
Hi @msk-repo-0x4d im trying to make a dll with libmodbus in c++ and then run that dll in c# but im facing a lot of errors could you pls help me tackle them? my email is: abhishekbade1310@gmail.com.