stephane/libmodbus #559: libmodbus 3.0.6   CRC error

wanljb, 2020-09-16:
i use libmodbus 3.0.6 to get data from RS485 senssors，but it always return "ERROR CRC received 110 != CRC calculated 9C93"！

    here is my code:

    modbus_t *modbus_ctx = modbus_new_rtu("/dev/ttymxc3", 9600, 'N', 8, 1);
    modbus_set_slave(modbus_ctx ,0x01);
    modbus_connect(modbus_ctx );
    modbus_set_response_timeout(modbus_ctx, &timeout);
    modbus_rtu_set_serial_mode(modbus_ctx,MODBUS_RTU_RS485);   // here also return -1 , modbus_strerror(errno) : Invalid argument
    while(isRun){
        u_int16_t dest[3];
        memset(dest, 0, sizeof(3));

        modbus_flush(modbus_ctx);
        modbus_read_registers(modbus_ctx, 0x01, 3, dest);     // here always return -1,modbus_strerror(errno):Invalid CRC
        sleep(1);
    }

karlp, 2020-09-16:
Does your parity match what the remote device is configured as? Does the baud rate match waht the remote device is configured as?  Are you _sure_ ? run in verbose mode, see what you're actually getting.  If you get errors from trying to set rs485, check whether your hardware supports that mode, or whether it's transparent, or any third paths.

wanljb, 2020-09-16:
> Does your parity match what the remote device is configured as? Does the baud rate match waht the remote device is configured as? Are you _sure_ ? run in verbose mode, see what you're actually [...]

yes , i am sure remote rs485 senssor is OK，i have used pc tools to connect it  ,it return the right data。

stephane, 2021-05-10:
The branch v3.0.x is not maintained anymore. Could you test again with latest version and run the code in debug mode to provide received data, please?