stephane/libmodbus #562: libmodbus on WSL

viktor-morin, 2020-10-15:
After trying to compile the code with several issue on WSL (Windows Subsystem for Linux),
I managed to get it to work follow following steps:

```sh
git clone https://github.com/stephane/libmodbus.git
cd libmodbus
libtoolize --force
autoheader
automake --force-missing --add-missing
autoconf
./configure
sudo make install

cd /etc/ld.so.conf.d/
nano modbus.conf
```

Then just close GNU nano (empty modbus.conf) by pressing **CTRL+X**

```sh
sudo ldconfig
```

Example code C (example.c)
```c
#include <stdio.h>
#include <unistd.h>
#include <stdlib.h>
#include <errno.h>
#include <modbus.h>

int main(void)
{
  int i;
  int s = -1;
  modbus_t *ctx;
  modbus_mapping_t *mb_mapping;
  
  ctx = modbus_new_tcp("127.0.0.1", 1502);
  //    modbus_set_debug(ctx, TRUE); 
  
  mb_mapping = modbus_mapping_new(0, 0, 500, 500);
  if (mb_mapping == NULL) {
    fprintf(stderr, "Failed to allocate the mapping: %s\n",
	    modbus_strerror(errno));
    modbus_free(ctx);
    return -1;
  }
  
  s = modbus_tcp_listen(ctx, 4);
  modbus_tcp_accept(ctx, &s);
  
  for (;;) {
    uint8_t query[MODBUS_TCP_MAX_ADU_LENGTH];
    int rc;
    
    printf("recive nxt \n");
    rc = modbus_receive(ctx, query);
    printf("SLAVE: regs[] =\t");
    for(i = 1; i != 11; i++) { // looks like 1..n index
      printf("%d ", mb_mapping->tab_registers[i]);
    }
    printf("\n");
    
    if (rc > 0) {
      /* rc is the query size */
      modbus_reply(ctx, query, rc, mb_mapping);
    } else if (rc == -1) {
      /* Connection closed by the client or error */
      break;
    }
  }
  
  printf("Quit the loop: %s\n", modbus_strerror(errno));
  
  if (s != -1) {
    close(s);
  }
  modbus_mapping_free(mb_mapping);
  modbus_close(ctx);
  modbus_free(ctx);
  
  return 0;
}
```

Compile running:
```sh
gcc example.c -o example -I/usr/local/include/modbus/ -lmodbus
```

closed 2020-10-15 by viktor-morin

chintanp, 2021-06-21:
@viktor-morin can you explain what does code sample do?

viktor-morin, 2021-06-22:
> @viktor-morin can you explain what does your code sample do?

The project itself is not relevant.
I had issue compile the lib on WSL, but manange to get it to work usinig the commands stated above.
I later on also realized the issue was related to the port used, which WSL didnt have default permission to.