stephane/libmodbus #563: Cannot compile libmodbus-3.0.8 on haiku64bit (r1b2)

mazbrili, 2020-10-22:
## libmodbus version
libmodbus 3.0.8
## OS and/or distribution
Haiku 64bit ( hrev54154+119)
## Environment
X86_64
## Description
-Problem cannot build libmodbus 3.0.8 on haiku64bit (r1b2)

## Steps to reproduce the behavior (commands or source code)
*to reproduce.
 *get the source version 3.0.8
 *run autogen.sh
 *run ./configure
 *run make

## Expected behaviour
compile successfully

## Actual behaviour
error make

## libmodbus output with debug mode enabled

 make[3]: Entering directory '/HPKG/haikuports/net-libs/libmodbus/work-3.0.8/sources/libmodbus-3.0.8/tests'
  CC       bandwidth-server-one.o
  CCLD     bandwidth-server-one
  CC       bandwidth-server-many-up.o
  CCLD     bandwidth-server-many-up
/boot/system/develop/tools/bin/../lib/gcc/x86_64-unknown-haiku/8.3.0/../../../../x86_64-unknown-haiku/bin/ld: bandwidth-server-many-up.o: in function `main':
/HPKG/haikuports/net-libs/libmodbus/work-3.0.8/sources/libmodbus-3.0.8/tests/bandwidth-server-many-up.c:104: undefined reference to `accept'
/boot/system/develop/tools/bin/../lib/gcc/x86_64-unknown-haiku/8.3.0/../../../../x86_64-unknown-haiku/bin/ld: /HPKG/haikuports/net-libs/libmodbus/work-3.0.8/sources/libmodbus-3.0.8/tests/bandwidth-server-many-up.c:114: undefined reference to `inet_ntoa'
collect2: error: ld returned 1 exit status
Makefile:429: recipe for target 'bandwidth-server-many-up' failed
make[3]: *** [bandwidth-server-many-up] Error 1
make[3]: Leaving directory '/HPKG/haikuports/net-libs/libmodbus/work-3.0.8/sources/libmodbus-3.0.8/tests'
Makefile:370: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/HPKG/haikuports/net-libs/libmodbus/work-3.0.8/sources/libmodbus-3.0.8/tests'
Makefile:466: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/HPKG/haikuports/net-libs/libmodbus/work-3.0.8/sources/libmodbus-3.0.8'
Makefile:373: recipe for target 'all' failed
make: *** [all] Error 2

mazbrili, 2020-10-23:
we can build successfully with add LIBS=-lnetwork  before configure.
we pass tcp test on haiku64

Begasus, 2020-10-31:
Builds fine with the current PR at haikuports for 32 bit Haiku (with gcc2 and gcc8) too

Begasus, 2020-11-04:
Adding next line to configure.ac solves the issue on using LIBS=-lnetwork
```diff
# Checks for library functions.
-AC_CHECK_FUNCS([accept4 getaddrinfo gettimeofday inet_ntoa select socket strerror strlcpy])
+AC_SEARCH_LIBS(inet_ntoa, network socket)
+AC_CHECK_FUNCS([accept4 getaddrinfo gettimeofday inet_ntoa select socket strerror strlcpy])
```
Checked with latest release and seems to be working fine

stephane, 2022-01-08:
Could we merge this change w/o issue for other platforms?

Begasus, 2022-01-09:
Changed the check a bit for Haiku like (for latest release: 3.1.7):

```diff
+# Check for Haiku's libnetwork library
+AC_SEARCH_LIBS(accept, network socket)
```

Checked a build on 32bit with both our compilers, also checked a build on Ubuntu, works fine.

With gcc2 we get an error on build time (but I guess we can strike that as broken as there is nothing using the gcc2 version).

```
/sources/libmodbus-3.1.7/tests/unit-test-client.c:32: badly punctuated parameter list in `#define'
```

For the checks I had to replace `killall` with `kill` (we don't have killall in Haiku) in: https://github.com/stephane/libmodbus/blob/1d758ca214bfb792c82c6d8e545c2e3e00fa05c0/tests/unit-tests.sh#L17

On both Haiku and Ubuntu it produces the same output:

```
make  check-TESTS
PASS: unit-tests.sh
============================================================================
Testsuite summary for libmodbus 3.1.7
============================================================================
# TOTAL: 1
# PASS:  1
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
# ERROR: 0
============================================================================
```

Begasus, 2022-01-09:
Hold this thought for a while, will get back on this later :)

closed 2022-02-21 by stephane