stephane/libmodbus #576: modbus ascii support

clustor, 2021-02-12:
I wanted to check if libmodbus officially supports ascii. I see no references to ascii support in the official documentation (https://libmodbus.org/docs/v3.1.6/) but I see ascii-support branch that has the relevant source code but hasn't been updated in a while.

jan-kardell, 2021-08-22:
It would be nice to have ASCII supported. I don't have need for it currently, but who known what the future holds...

One usual argument is that it's not needed, as all devices support RTU, and few devices support ASCII. But that's not entirely true, RTU has very stringent timing requirements while ASCII is much more  relaxed. So thare are some situations, e.g. when connecting through some kind of gateway, that ASCII is needed because the timing requirements of RTU cannot be met.

asperg, 2022-10-12:
I have only one model from the HT11 1-3KVA line, this model supports modbus ascii and STN protocols. It was possible to obtain documentation only for the modbus protocol. Modbus ascii is enabled by default. I also made attempts to connect to UPS via modbus rtu, but they were unsuccessful.