stephane/libmodbus #578: Configure error compiling for Windows 10

effemmeffe, 2021-02-26:
## libmodbus version

Commit: 1d758ca214bfb792c82c6d8e545c2e3e00fa05c0 [1d758ca]

## OS and/or distribution

Windows 10

## Environment

64 bit

## Description

Trying to compile libmodbus on a Windows 10 machine.
I have the latest MSYS2 installed.
I cloned the sources from git://github.com/stephane/libmodbus

In a MSYS2 command line I run the autogen.sh and then the configure scripts, but it fails.
Following the result of the two commands.
Any help appreciated.

## Expected behaviour

libmodbus to compile

## Actual behaviour

./configure errors

## Steps to reproduce the behavior (commands or source code)

```
effemmeffe@PC10-effemmeffe MSYS /C/Progetti/libmodbus
$ ./autogen.sh
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
libtoolize: linking file 'build-aux/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:33: warning: The macro `AC_PROG_CC_STDC' is obsolete.
configure.ac:33: You should run autoupdate.
../autoconf-2.71/lib/autoconf/c.m4:1666: AC_PROG_CC_STDC is expanded from...
configure.ac:33: the top level
): you should use literalsAC_CHECK_HEADERS(\
../autoconf-2.71/lib/autoconf/headers.m4:217: AC_CHECK_HEADERS is expanded from...
configure.ac:77: the top level
configure.ac:33: installing 'build-aux/compile'
configure.ac:56: installing 'build-aux/config.guess'
configure.ac:56: installing 'build-aux/config.sub'
configure.ac:32: installing 'build-aux/install-sh'
configure.ac:32: installing 'build-aux/missing'
src/Makefile.am: installing 'build-aux/depcomp'
parallel-tests: installing 'build-aux/test-driver'

------------------------------------------------------
Initialized build system. You can now run ./configure
------------------------------------------------------

effemmeffe@PC10-effemmeffe MSYS /C/Progetti/libmodbus
$ ./configure
configure: loading site script /etc/config.site
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for wchar.h... yes
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking whether _XOPEN_SOURCE should be defined... no
checking for special C compiler options needed for large files... no
checking for _FILE_OFFSET_BITS value needed for large files... no
checking whether make supports nested variables... (cached) yes
checking build system type... x86_64-pc-msys
checking host system type... x86_64-pc-msys
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/x86_64-pc-msys/bin/ld.exe
checking if the linker (/usr/x86_64-pc-msys/bin/ld.exe) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -pR
checking the maximum length of command line arguments... 8192
checking how to convert x86_64-pc-msys file names to x86_64-pc-msys format... func_convert_file_noop
checking how to convert x86_64-pc-msys file names to toolchain format... func_convert_file_noop
checking for /usr/x86_64-pc-msys/bin/ld.exe option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... file_magic ^x86 archive import|^x86 DLL
checking for dlltool... dlltool
checking how to associate runtime and link libraries... func_cygming_dll_for_implib
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for dlfcn.h... yes
checking for as... as
checking for dlltool... (cached) dlltool
checking for objdump... (cached) objdump
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -DDLL_EXPORT -DPIC
checking if gcc PIC flag -DDLL_EXPORT -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/x86_64-pc-msys/bin/ld.exe) supports shared libraries... yes
checking whether -lc should be explicitly linked in... yes
checking dynamic linker characteristics... Win32 ld.exe
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... no
./configure: line 13872: syntax error near unexpected token `as_ac_Header=`printf "%s\n" "ac_cv_header_$ac_header" | $as_tr_sh`'
./configure: line 13872: `  as_ac_Header=`printf "%s\n" "ac_cv_header_$ac_header" | $as_tr_sh`'

effemmeffe@PC10-effemmeffe MSYS /C/Progetti/libmodbus
$

```

dreamflow, 2021-02-26:
on a Win10-64bit computer within MSYS2_64bit
i've installed with the pacman-command the following packages

`pacman -S autoconf automake-wrapper libtool make mingw-w64-i686-gcc`

and then could compile with 32bit gcc the 32bit dll of libmodbus .

with the 64bit mingw-gcc package installed in MSYS2_64bit ,
its also possible to compile a 64bit dll of libmodbus .
but as reported in https://github.com/stephane/libmodbus/issues/532 , the RTU part didn't work in the 64bit dll .

effemmeffe, 2021-02-26:
How do I choose gcc 32 rather than gcc 64?
Also, I want to use the dll in a C program, I think I would need a .lib and the headers, can you tell me how to compile for that?

dreamflow, 2021-02-26:
in MSYS2_64bit run mingw32.exe , then within the
MINGW32-terminal the "gcc"-command is 32bit .

it could be , that the .lib file is also created during
the build , but i'm not sure . i don't need it .
i do only dynamic calls on the libmodbus dll .

effemmeffe, 2021-02-26:
I'm new to this type of programming, I have a background in embedded programming.
You're saying that I can use only the dll and the headers files and only call the functions in my code? How do I tell to the linker to use the dll? 
I'm planning to compile with gcc in code::blocks.

dreamflow, 2021-02-26:
i'm doing dynamic calls to the libmodbus dll from LabVIEW ,
that has the capability of calling dll's dynamically .
technically it's similar to calling a dll from python with ctypes or CFFI .

doing such indirect calls to a C library like libmodbus from C is not needed
and probably more complicated than just using the C library libmodbus
directly form a C program .
i hope that other users here could help you with how to use libmodbus from C .

iw2lsi, 2021-12-02:
Hi all,

  I'm also experiencing the same problem... and I do not understand if/how it has been resolved.

I'm using libmodbus 3.1.6 (tested both with git and zip archive) and msys2-x86_64-20210725 /msys2-x86_64-20211130, both downloaded from https://www.msys2.org and installed as described in their homepage.

Below is the log of the MINGW32 console...

   thanks for any hints...

              Giampaolo

$ ./autogen.sh
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
libtoolize: linking file 'build-aux/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:33: warning: The macro `AC_PROG_CC_STDC' is obsolete.
configure.ac:33: You should run autoupdate.
../autoconf-2.71/lib/autoconf/c.m4:1666: AC_PROG_CC_STDC is expanded from...
configure.ac:33: the top level
): you should use literalsAC_CHECK_HEADERS(\
../autoconf-2.71/lib/autoconf/headers.m4:217: AC_CHECK_HEADERS is expanded from...
configure.ac:77: the top level
configure.ac:33: installing 'build-aux/compile'
configure.ac:32: installing 'build-aux/missing'
src/Makefile.am: installing 'build-aux/depcomp'

------------------------------------------------------
Initialized build system. You can now run ./configure
------------------------------------------------------

iw2ls@DESKTOP-BU09PBT MINGW32 /c/home/iw2lsi/projects/libmodbus
$ ./configure
configure: loading site script /etc/config.site
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for stdio.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for strings.h... yes
checking for sys/stat.h... yes
checking for sys/types.h... yes
checking for unistd.h... yes
checking for wchar.h... yes
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
[... 30 lines cut ...]
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... no
checking if : is a manifest tool... no
checking for dlfcn.h... yes
checking for as... as
checking for dlltool... (cached) dlltool
checking for objdump... (cached) objdump
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -DDLL_EXPORT -DPIC
checking if gcc PIC flag -DDLL_EXPORT -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/x86_64-pc-msys/bin/ld.exe) supports shared libraries... yes
checking whether -lc should be explicitly linked in... yes
checking dynamic linker characteristics... Win32 ld.exe
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... no
./configure: line 13872: syntax error near unexpected token `as_ac_Header=`printf "%s\n" "ac_cv_header_$ac_header" | $as_tr_sh`'
./configure: line 13872: `  as_ac_Header=`printf "%s\n" "ac_cv_header_$ac_header" | $as_tr_sh`'

iw2lsi, 2021-12-02:
Hi all...

   my bad... I've cloned the project using git from within the MINGW32 shell instead of doing it using tortoise-git on a "generic" path and now it's working...

      Giampaolo

iw2lsi, 2021-12-03:
Hi,

   not sure it works but at least it compiles... 

 copy:
   C:\Windows\SysWOW64\ws2_32.dll
to:
   C:\msys2-x86_64-20211130\mingw32\i686-w64-mingw32\lib\libws2_32.dll 

and copy:
   C:\Windows\SysWOW64\wsock32.dll
to:
   C:\msys2-x86_64-20211130\mingw32\i686-w64-mingw32\lib\libwsock32.dll

    Giampaolo

modem-man-gmx, 2022-11-29:
> on a Win10-64bit computer within MSYS2_64bit i've installed with pacman the following packages
> `pacman -S autoconf automake-wrapper libtool make mingw-w64-i686-gcc`

Hey, you saved my day! 
I missed to install autoconf and automake-wrapper and got strange error messages.

modem-man-gmx, 2022-11-29:
> How do I choose gcc 32 rather than gcc 64? Also, I want to use the dll in a C program, I think I would need a .lib and the headers, can you tell me how to compile for that?

I installed MSY2 in 64 bit version, and in Win10 I got multiple icons/quicklinks with each one stands for one compiler/environment:

- magenta/violet for general MSYS2 shell (perhaps with cygwin?)
- grey for MINGW32
- blue for MING64
- brown for UCRT64 (successor for MSVCRT)
- orange for CLANG64

when I need to write 32 bit stuff with pure make, I just open the grey icon's shell, g++ is then:

```
$ which g++
/mingw32/bin/g++
```
while the blue icon creates a bash with:
```
$ which g++
/mingw64/bin/g++
```

(Except when working from Codelite, which needs some of its own tools in path and can scan and offer all the MSYS2 compiler frontends, there I just start Codelite and select the compiler from the project properties)

HTH,
Mo

modem-man-gmx, 2022-11-29:
Code::Blocks and CodeLite are quite related, so my experience of better *not* calling codelite from within MSYS2 shell could also apply to Code::Blocks? I had:

- mkdir /p will only work if codelite is in $PATH, since this tool is there
- UnitTest++ seems to only work if codelite was invoked from start menu or CMD shell, never found the reason
- CodeCompletion ... same, never found the root cause

On DLL ... 
why do you want use it via DLL? The libmodbus isn't that large, to save storage by sharing same DLL with different modbus tools. You could link it statically. The LGPL 2.1 makes no difference here.

BobinInk, 2023-01-16:
Sorry for the necro

I have the same issue
I installed all the packages listed by @dreamflow, tried with the 32bit mingw and the 64bit. I still get the same error as @effemmeffe

dreamflow, 2023-01-16:
@BobinInk :
for my last successful libmodbus.dll 64bit compilation ,
i've installed into MSYS2_64bit the needed packages via
```
pacman -S autoconf automake-wrapper libtool make mingw-w64-i686-gcc mingw-w64-x86_64-gcc mingw-w64-ucrt-x86_64-gcc
```
in MSYS2_64bit i've started `ucrt64.exe` and within this
ucrt64-terminal i've performed the compilation steps on the
downloaded and extracted libmodbus-source-github-master-branch-zip via
```
./autogen.sh
./configure
make
```
that resulted in a functioning 64bit libmodbus.dll .

znstj, 2023-03-02:
Build error in Win 10/Win 7 MinGW 32/64 with MSYS2
How build  libmodbus.dll 32bit？

znstj, 2023-03-02:
> Build error in Win 10/Win 7 MinGW 32/64 with MSYS2 How build libmodbus.dll 32bit？

OK！
Download all ，and checkout v3.1.10，build error 
But just download tag v3.1.10,build correctly！