stephane/libmodbus #592: Does this lib make assumption that it always used on little endian host?

dsvost, 2021-07-18:
If I correctly understand the following lines swap received bytes. If it is it then yes, the lib assume that we are on a little-endian system now. I think it has sense to add in README.md explicit notice as for it, like "the implementation doesn't aware of actual host endianness and make assumption that it is little-endian always". And then it will be clear, if we use it on big-endian system then we need swap bytes in returned words one more time on lib API caller side.

https://github.com/stephane/libmodbus/blob/31f779185d3d2a7b212b508ea321f4e449f3de85/src/modbus.c#L1178

karlp, 2021-08-03:
fwiw, I use this heavily on ath79, which is a big endian device, with no problems.

dsvost, 2021-08-03:
Yes, thank you. I think I was too tired when did open this issue, cause as I look it now it seems it doesn't swap the bytes and just put bytes in order. Bits shift operations doesn't assume any endianess. Shift left is always shift to higher order position and it doesn't depend on endianess. So, I was wrong in original question.

closed 2021-08-03 by dsvost